simpler-env/SimplerEnv #7: [error] GLFW error: X11: Failed to open display

DelinQu, 2024-06-29:
Hi,
Thanks for your meaningful work! I just try to evaluate `octo` and custom my new policy. The environment is set in a A100 cluster, but got this error:
```bash
Control mode:  arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos
[2024-06-30 06:12:53.528] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2024-06-30 06:12:53.528] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/simpler_env/main_inference.py", line 68, in <module>
    success_arr = maniskill2_evaluator(model, args)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 215, in maniskill2_evaluator
    success_arr.append(run_maniskill2_eval_single_episode(obj_episode_id=obj_episode_id, **kwargs))
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 60, in run_maniskill2_eval_single_episode
    env = build_maniskill2_env(
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/simpler_env/utils/env/env_builder.py", line 16, in build_maniskill2_env
    env = gym.make(env_name, **kwargs)
  File "/mnt/petrelfs/qudelin/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 370, in __init__
    super().__init__(
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 351, in __init__
    super().__init__(
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 162, in __init__
    super().__init__(**kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/move_near_in_scene.py", line 58, in __init__
    super().__init__(**kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/base_env.py", line 134, in __init__
    super().__init__(**kwargs)
  File "/mnt/hwfile/optimal/qudelin/Gripper2024/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: Cannot find a suitable rendering device
```
Does this simulator support Desktop PC only? Could you provide any advice, Many thanks! 🤗

xuanlinli17, 2024-06-30:
Hi, please follow https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan  and ensure that the required nvidia json files etc exist, and nvidia-driver version is at least 535.

DelinQu, 2024-07-01:
Thanks, [xuanlinli17](https://github.com/xuanlinli17); I carefully followed all the instructions but still failed in an A100 Slurm cluster (even in an apptainner). Finally, I successfully deployed SimplerEnv on a local PC with 4090 GPU and evaluated the [Openvla](https://github.com/openvla/openvla) policy in this benchmark.

closed 2024-07-01 by DelinQu

HuFY-dev, 2024-07-04:
Hi @DelinQu, I hope you're doing well. I'm also interested in evaluating OpenVLA on SimplerEnv. Would it be possible for you to share your evaluation scripts? I would greatly appreciate your assistance. Thank you in advance for your help!

hongzhitao, 2024-07-19:
> 嗨，我希望你一切顺利。我也有兴趣在 SimplerEnv 上评估 OpenVLA。您能否分享您的评估脚本？我将非常感谢您的帮助。提前感谢您的帮助！

I'm also doing the same job, can we talk about it？