simpler-env/SimplerEnv #11: Controller difference for different envs

HuFY-dev, 2024-07-06:
Hi, thanks for the great work! I'm playing around with your environments and found that both RT1 and Octo seems to be capable of only the `google_robot` tasks but not the `widowx` tasks.

Further, I noticed that `google_robot` environments use the `arm_pd_ee_delta_pose` + `gripper_pd_joint_target_delta_pos` controllers while `widowx` environments use the `arm_pd_ee_target_delta_pose` + `gripper_pd_joint_pos` controllers. Notice that both the arm and gripper controllers are different. However, in your model wrappers, you seem to be treating the `world_vector` and `rot_axangle` the same way regardless of the difference in the controller. I wonder if that's causing the models to fail `widowx` tasks

FYI, I got the controllers using `env.unwrapped.control_mode`. More details on controllers can be found [here](https://maniskill.readthedocs.io/en/latest/user_guide/concepts/controllers.html#arm-controllers)

xuanlinli17, 2024-07-07:
Hi,

Octo models do not fail on the widowx tasks. The controller we use are here: https://github.com/simpler-env/SimplerEnv/blob/main/simpler_env/utils/env/env_builder.py .

HuFY-dev, 2024-07-07:
Thanks, let me look into that!

HuFY-dev, 2024-07-07:
Also, why there's no difference in handling the different arm controllers in the model wrappers? Seems like only the gripper actions are handled differently. 

In terms of gripper actions, can you explain why you set [`self.sticky_gripper_num_repeat = 15`](https://github.com/simpler-env/SimplerEnv/blob/bcf62960dc9d95501cf36dc8c64d2128e9b10e09/simpler_env/policies/octo/octo_model.py#L40C18-L40C48) for google and 1 for bridge? I feel like it came from nowhere

xuanlinli17, 2024-07-07:
The `sticky_gripper_num_repeat` was set only for Octo models to match the real eval. OpenVLA likely uses a different setup, so need to connect to the authors to verify how they implement the sticky actions in real eval.

HuFY-dev, 2024-07-07:
One failure case of OpenVLA: 

[image]

[Here](https://github.com/openvla/openvla/issues/3) are some explanations from the OpenVLA authors, but I don't think they are clear enough.

Can you share where you found the `sticky_gripper_num_repeat` in the original setting from the authors of Octo?

xuanlinli17, 2024-07-07:
This is almost surely an implementation issue. Looks like the rotation orderings might be wrong for Bridge envs. OpenVLA might output ypr instead of rpy for Bridge. That is, while for Octo, `roll, pitch, yaw = action_rotation_delta`, for OpenVLA, it might be the case that `yaw, pitch, roll = action_rotation_delta`.

`sticky_gripper_num_repeat` is not in the public Octo repo. I communicated with the authors directly and referenced their real eval implementations.

HuFY-dev, 2024-07-07:
I tried all arrangements and neither worked. The task is to pick up the spoon, and the gripper should move forward in order to do that. However, every time it always goes straight down without moving forward, which is very weird.

Toradus, 2024-07-10:
Have you tested with other tasks?
Im currently training my own custom model and try to make a eval on SimplerEnv, which works very nice for Fractal but also struggles alot with training on the Octo provided Bridge_v2 and evaluating on the spoon task.
I also get the same results of the robot driving into the table for the spoon task, but for the carrot task it is at least aiming at the right target. Note that this could easily be a model problem on my end, but its interesting that the OpenVLA eval looks about the same as my eval. I copy pasted the [OctoInference provided by Simpler](https://github.com/simpler-env/SimplerEnv/blob/bcf62960dc9d95501cf36dc8c64d2128e9b10e09/simpler_env/policies/octo/octo_model.py#L16) and adjusted it for my model to work. Did OpenVLA do the same?

https://github.com/simpler-env/SimplerEnv/assets/23532458/05db04a4-f943-4652-aa08-241c3fbd3d90

https://github.com/simpler-env/SimplerEnv/assets/23532458/91ac59cc-ece2-48bc-9c63-2dd54aeb5700

HuFY-dev, 2024-07-10:
Yes mine model wrapper was also modified from the Octo code. However, from my understanding, the main part of that code is processing the rotation from rpy to axis angles and perform the sticky gripper logic, but it seems what's going wrong is the translation. Here are some carrot examples:

[image]

[image]