simpler-env/SimplerEnv #12: Segmentation fault when computing IK

ZhangHanbo, 2024-07-09:
System: UBUNTU 20.04
Python: 3.10.14

All codes are set up with the introduction in README. _vulkaninfo_ and _python -m sapien.example.offscreen_ works fine, seems that there are no rendering issues.

Segmentation fault is located at:
ManiSkill2_real2sim/mani_skill2_real2sim/agents/controllers/pd_ee_pose.py line 61:
result, success, error = self.pmodel.compute_inverse_kinematics(
            self.ee_link_idx,
            target_pose,
            initial_qpos=self.articulation.get_qpos(),
            active_qmask=self.qmask,
            max_iterations=max_iterations,
        )

How can I solve this problem?

xuanlinli17, 2024-07-11:
Another person has pointed out that pinocchio encounters issue when numpy>=2.0. 
Installing numpy==1.24.4         (numpy<2.0) will resolve this issue.

closed 2024-07-11 by xuanlinli17

LukeLIN-web, 2025-05-10:
> Another person [#6](https://github.com/simpler-env/SimplerEnv/issues/6) has pointed out that pinocchio encounters issue when numpy>=2.0. Installing numpy==1.24.4 (numpy<2.0) will resolve this issue. I've updated the readme.

Thank you . It works! numpy 1.26.4  leads to Segmentation fault