simpler-env/SimplerEnv #13: Pick coke can variants.

HuFY-dev, 2024-07-09:
I saw in your eval scripts for pick coke can: [`declare -a coke_can_options_arr=("lr_switch=True" "upright=True" "laid_vertically=True")`](https://github.com/simpler-env/SimplerEnv/blob/bcf62960dc9d95501cf36dc8c64d2128e9b10e09/scripts/rt1_pick_coke_can_visual_matching.sh#L11).

In Table IV in your paper, there's `Horizontal Laying`, `Vertical Laying`, `Standing` three variants. Does `lr_switch=True` correspond to the `Horizontal Laying` variant? If so, does that mean you didn't evaluate the default setting where `"lr_switch=False, upright=False, laid_vertically=False"`?

xuanlinli17, 2024-07-09:
"lr_switch" is coke can oriented by 180 degrees to match real eval's "horizontally laying" setting.

If you set "lr_switch=False, upright=False, laid_vertically=False" args to False, then the coke can will be oriented randomly. There is no "default setting". The purpose of our experiments is to draw correlation with real-world evaluations under a similar setup, and 3 default orientations (horizontal / vertical laying, standing) with a "grid sweep" over coke can positions already covers many variations.

HuFY-dev, 2024-07-10:
Thanks for the clarification!

closed 2024-07-10 by HuFY-dev

erjiaxiao, 2024-08-27:
Hello @xuanlinli17, I wonder is it possible to change the position of the Coke can in the simulator?

xuanlinli17, 2024-08-27:
For xy positions, yes (see example scripts)