simpler-env/SimplerEnv #15: [svulkan2] [error] GLFW error: X11: Failed to open display

junh0cho, 2024-07-12:
Hi, I am having same issue with https://github.com/simpler-env/SimplerEnv/issues/7 , but could not solve it.
My system is RTX3070, driver 535.

I also checked https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan, and made sure all jsons are there.

```
(simpler_env) juny@user-Nuvo-8208GC-Series:~/workspace_jh/SimplerEnv$ cat /usr/share/vulkan/icd.d/nvidia_icd.json
{

    "file_format_version" : "1.0.0",

    "ICD": {

        "library_path": "libGLX_nvidia.so.0",

        "api_version" : "1.2.155"

    }

}
(simpler_env) juny@user-Nuvo-8208GC-Series:~/workspace_jh/SimplerEnv$ cat /usr/share/glvnd/egl_vendor.d/10_nvidia.json
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
(simpler_env) juny@user-Nuvo-8208GC-Series:~/workspace_jh/SimplerEnv$ cat /etc/vulkan/implicit_layer.d/nvidia_layers.json
{
    "file_format_version" : "1.0.0",
    "layer": {
        "name": "VK_LAYER_NV_optimus",
        "type": "INSTANCE",
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.3.242",
        "implementation_version" : "1",
        "description" : "NVIDIA Optimus layer",
        "functions": {
            "vkGetInstanceProcAddr": "vk_optimusGetInstanceProcAddr",
            "vkGetDeviceProcAddr": "vk_optimusGetDeviceProcAddr"
        },
        "enable_environment": {
            "__NV_PRIME_RENDER_OFFLOAD": "1"
        },
        "disable_environment": {
            "DISABLE_LAYER_NV_OPTIMUS_1": ""
        }
    }
}
```

but following error still comes out
```
(simpler_env) juny@user-Nuvo-8208GC-Series:~/workspace_jh/SimplerEnv$ bash ./scripts/octo_put_in_drawer_visual_matching.sh
...
Fetching 8 files: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 8/8 [00:00<00:00, 13117.45it/s]
Control mode:  arm_pd_ee_delta_pose_align_interpolate_by_planner_gripper_pd_joint_target_delta_pos_interpolate_by_planner
[2024-07-13 02:51:40.243] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2024-07-13 02:51:40.243] [svulkan2] [warning] Continue without GLFW.
/home/juny/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning:
...
194 {'elapsed_steps': 195, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
195 {'elapsed_steps': 196, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
196 {'elapsed_steps': 197, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
197 {'elapsed_steps': 198, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
198 {'elapsed_steps': 199, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
199 {'elapsed_steps': 200, 'success': False, 'episode_stats': OrderedDict([('qpos', 3.3117403e-10), ('is_drawer_open', False), ('has_contact', 0)])}
[2024-07-13 02:54:53.640] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2024-07-13 02:54:53.640] [svulkan2] [warning] Continue without GLFW.
open top drawer
0 {'elapsed_steps': 1, 'success': False, 'episode_stats': OrderedDict([('qpos', 4.321821e-15), ('is_drawer_open', False), ('has_contact', 0)])}
1 {'elapsed_steps': 2, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
2 {'elapsed_steps': 3, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
3 {'elapsed_steps': 4, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
4 {'elapsed_steps': 5, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
5 {'elapsed_steps': 6, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
6 {'elapsed_steps': 7, 'success': False, 'episode_stats': OrderedDict([('qpos', 5.2316766e-15), ('is_drawer_open', False), ('has_contact', 0)])}
...
```

StoneT2000, 2024-07-12:
Does a video get generated? Sometimes when this shows up it is fine, it's just trying to look for a display to open a GUI but did not find one (Does not mean it cannot render videos).

Did you get the same error in #7 at the very end where it says it cannot find a rendering device? If so can you run
```
nvidia-smi
vulkaninfo
```
and share the results here?

xuanlinli17, 2024-07-12:
You can ignore these errors as long as the policy evaluation can run.
```
[2024-07-13 02:51:40.243] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2024-07-13 02:51:40.243] [svulkan2] [warning] Continue without GLFW.
```

junh0cho, 2024-07-15:
I see, results are saved in `./SimplerEnv/resuls/octo-base` and `./SimplerEnv/resuls/rt_1_x_tf_trained_for_002272480_step`.
Those mp4 videos are playable with vlc. 
Thanks!

closed 2024-07-15 by junh0cho

wugh, 2024-08-03:
> [2024-07-13 02:51:40.243] [svulkan2] [error] GLFW error: X11: Failed to open display

hi, xuanlinli
so what is this error mean? I google it but can not find how to fix it.
I also have this error, but I can evaluate the policy.

StoneT2000, 2024-08-03:
This usually means there isn't a display / monitor the server is connected to. You can ignore it.

wugh, 2024-08-04:
> This usually means there isn't a display / monitor the server is connected to. You can ignore it.

OK, but I run the code on local machine