simpler-env/SimplerEnv #18: Can it run on a Linux system with A100 GPUs?

RZFan525, 2024-07-25:


xuanlinli17, 2024-07-25:
Yes. Please follow the instructions in readme and troubleshooting. Though, rendering for the drawer tasks will be slow due to the use of ray tracing.

RZFan525, 2024-07-26:
Thank you for your reply. However, I encountered the same error as [#7](https://github.com/openvla/openvla/issues/7). And, when I install vulkan-utils with `sudo apt-get install vulkan-utils`, an error appears:
`
The package vulkan-utils could not be located
`
I don't have any computers with RTX GPUs, how can I run it?

RZFan525, 2024-07-26:
I have followed [https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan) to add three json files, but it does not work.

xuanlinli17, 2024-07-27:
Did you `sudo apt update` and `vulkan-utils` is still not found?

RZFan525, 2024-07-28:
Thank you for your reply. I have tried it, and it can be installed successfully. However, the same error has appeared.

And, I found that `vulkaninfo` works without `/usr/share/vulkan/icd.d/nvidia_icd.json`, `/usr/share/glvnd/egl_vendor.d/10_nvidia.json`, and `/etc/vulkan/implicit_layer.d/nvidia_layers.json`. But, when I follow https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan to manually add these three files, `vulkaninfo` doesn't work with the error `ERROR_OUT_OF_HOST_MEMORY`.

Anyway, the following error always appears whether the `vulkaninfo` can work or not.

```
[2024-07-28 11:58:15.019] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2024-07-28 11:58:15.019] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv-OpenVLA/test.py", line 4, in <module>
    env = simpler_env.make('google_robot_pick_coke_can')
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv-OpenVLA/simpler_env/__init__.py", line 78, in make
    env = gym.make(env_name, obs_mode="rgbd", **kwargs)
  File "/cpfs01/user/liupengfei/rzfan/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/grasp_single_in_scene.py", line 630, in __init__
    super().__init__(**kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/grasp_single_in_scene.py", line 540, in __init__
    super().__init__(**kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/grasp_single_in_scene.py", line 64, in __init__
    super().__init__(**kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/base_env.py", line 134, in __init__
    super().__init__(**kwargs)
  File "/cpfs01/user/liupengfei/rzfan/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed
```

RZFan525, 2024-07-28:
I don't know how to run it :(

I have tried three different servers with A100 GPUs which encounter the same error. :(

xuanlinli17, 2024-07-29:
Are you setting cuda devices properly? Also ensure that nvidia-driver version is above 530.

RZFan525, 2024-07-30:
Thank you for getting back to me.

The servers I used are in a docker and I changed to another server, which makes it work.

However, I encountered another error which is attributed to the lack of display.

```
RuntimeError: Create window failed: context is not created with present support
```

Do you have any suggestions to help me observe the environment and the process of action?

xuanlinli17, 2024-07-30:
Inside docker, you might want to port the (fake) display (e.g., `sudo X :0 &`) in the main bash to the docker container

However, the SIMPLER environments shouldn't create a window unless you are visualizing robots.

RZFan525, 2024-07-30:
I'm new in robotics, so I want to visualize the simulation environment to help me understand deeply. Maybe, it's better to output a video.

xuanlinli17, 2024-07-30:
The evaluation videos are automatically saved.

RZFan525, 2024-07-30:
Thank you! 

I can run the scripts `scripts/openvla_bridge.sh`, but it suddenly reports an error after running for a while.

[image]

xli-bdai, 2024-07-30:
if you create 2 envs in ipython in a row, does the error occur?

xuanlinli17, 2024-07-30:
If you consecutively create 2 environments in ipython, does it still report an error?

RZFan525, 2024-07-31:
When I create 2 environments, it can work. But there is a warning:

```
[2024-07-31 03:20:06.870] [svulkan2] [warning] A second renderer will share the same internal context with the first one. Arguments passed to constructor will be ignored.
```

RZFan525, 2024-07-31:
i don't know why. But, I also try `SimplerEnv-OpenVLA/scripts/openvla_drawer_variant_agg.sh` It's successful to output the average success

[image]

> Thank you!
> 
> I can run the scripts `scripts/openvla_bridge.sh`, but it suddenly reports an error after running for a while.
> 
> <img alt="image" width="1517" src="https://private-user-images.githubusercontent.com/41265224/353376196-654cd336-1e10-49da-8143-b7345f767951.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzOTUwOTgsIm5iZiI6MTcyMjM5NDc5OCwicGF0aCI6Ii80MTI2NTIyNC8zNTMzNzYxOTYtNjU0Y2QzMzYtMWUxMC00OWRhLTgxNDMtYjczNDVmNzY3OTUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMxVDAyNTk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkMTcxM2ExZDYzZTQ3OWIwYTQzZDM4NjJlMmQyODhhZDQ4Y2NhYTVlOGVhMjIwNjAwZTFiMmIwNWNlYWRlYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JgrEZYDTBw-_XhQlEvxvyXk_7V5fawP8Z59kmbT-qAo">

RZFan525, 2024-08-01:
I find that the error appears when the obj_episode_id is 11 in any scripts that define obj-variation-mode as the episode.
[image]

xuanlinli17, 2024-08-01:
That's strange; episode 11 doesn't introduce new objects.

RZFan525, 2024-08-02:
Could you give me some instructions on how to debug?  Thank you very much!!

xuanlinli17, 2024-08-02:
I actually don't know... and sorry that I don't have much bandwidth at the moment to look closely.

RZFan525, 2024-08-02:
Ok. Thank you for your reply.

xuanlinli17, 2024-08-02:
Also you might create fake display like `sudo X :0 &; export DISPLAY=:0` or `xvfb-run -a {script}`, to see if it works.

RZFan525, 2024-08-03:
Thank you. After trying this command, I found it cannot work. The error is the same. I don't know why.

COST-97, 2024-09-18:
Hello: 

> I don't know how to run it :(
> 
> I have tried three different servers with A100 GPUs which encounter the same error. :(

Same error in A100 GPU.
"libGLX_nvidia.so.0" does not exist in the A100.

Does anyone have an updated solution?
 Thanks a lot!

xuanlinli17, 2024-09-18:
> Hello:
> 
> > I don't know how to run it :(
> > I have tried three different servers with A100 GPUs which encounter the same error. :(
> 
> Same error in A100 GPU. "libGLX_nvidia.so.0" does not exist in the A100.
> 
> Does anyone have an updated solution? Thanks a lot!

Could you try the troubleshooting section in readme?

Akila-Ayanthi, 2025-01-12:
Same error in A100 GPU.
"libGLX_nvidia.so.0" does not exist in the A100.

Does anyone have an updated solution?
Thanks a lot!

LukeLIN-web, 2025-05-02:
> Did you `sudo apt update` and `vulkan-utils` is still not found since it's ubuntu 22.04?
> 
> Try `sudo apt install vulkan-tools`

> Thank you for your reply. I have tried it, and it can be installed successfully. However, the same error has appeared.
>
> And, I found that `vulkaninfo` works without [...]

I tried this as well, I have  different error,  it hang when create env https://github.com/simpler-env/SimplerEnv/issues/89

LukeLIN-web, 2025-05-02:
> I find that the error appears when the obj_episode_id is 11 in any scripts that define obj-variation-mode as the episode. <img alt="image" width="1420" src="https://private-user-images.githubusercontent.com/41265224/354172199-d419d501-d9b7-4b4c-8a30-82adeac2cdca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDYxNDg4ODQsIm5iZiI6MTc0NjE0ODU4NCwicGF0aCI6Ii80MTI2NTIyNC8zNTQxNzIxOTktZDQxOWQ1MDEtZDliNy00YjRjLThhMzAtODJhZGVhYzJjZGNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTAyVDAxMTYyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYwOGU5YTQzMmJiODM3NDcxNzVkODkwZDZiNTZhZWMyYzBmMTExOWIxNjk2YTAwMzg1MzhiMjRhOTA1YWI0NWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QUgAQWUaoYnPSN-ld4p7BWUlE11b0TCDFTjEyiWsGmU">

I think this is because it can only create 11 env maxmium , don't know where define it. I also meet this problem .

LukeLIN-web, 2025-05-02:
> Same error in A100 GPU. "libGLX_nvidia.so.0" does not exist in the A100.
> 
> Does anyone have an updated solution? Thanks a lot!

refer https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan