simpler-env/SimplerEnv #20: RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed

yxchng, 2024-08-02:
```
[2024-08-02 11:36:11.730] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2024-08-02 11:36:11.731] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/home/user/SimplerEnv/simpler_env/main_inference.py", line 60, in <module>
    success_arr = maniskill2_evaluator(model, args)
  File "/home/user/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 215, in maniskill2_evaluator
    success_arr.append(run_maniskill2_eval_single_episode(obj_episode_id=obj_episode_id, **kwargs))
  File "/home/user/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 60, in run_maniskill2_eval_single_episode
    env = build_maniskill2_env(
  File "/home/user/SimplerEnv/simpler_env/utils/env/env_builder.py", line 16, in build_maniskill2_env
    env = gym.make(env_name, **kwargs)
  File "/home/user/cache/conda/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 310, in __init__
    super().__init__(
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 162, in __init__
    super().__init__(**kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/move_near_in_scene.py", line 58, in __init__
    super().__init__(**kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/base_env.py", line 134, in __init__
    super().__init__(**kwargs)
  File "/home/user/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

I am meeting this problem when trying to run octo_bridge.sh script. How can I resolve this problem? Is a display required?

xuanlinli17, 2024-08-02:
Try the link in troubleshooting section and ensure 3 json files exist. Ensure numpy < 2.0. I don't think display is required.

xuanlinli17, 2024-08-02:
If above doesn't work, you can create a fake display using `sudo X :0 &` or sth like `xvfb-run -a {script}`.

closed 2024-08-13 by xuanlinli17