simpler-env/SimplerEnv #21: What's the difference between simplerenv and maniskill?

yxchng, 2024-08-02:


xuanlinli17, 2024-08-02:
SIMPLER environments use the SAPIEN simulator and the ManiSkill2 base environment. The individual environments inherit the ManiSkill2 base environment (and thus follow ManiSkill2's convention of controller, robot setup, etc).

@StoneT2000 is migrating the environment to SAPIEN 3 & ManiSkill3. The new environments will support GPU-parallelization.

StoneT2000, 2024-08-02:
@xuanlinli17 is spot on here. 

There is a larger project in ManiSkill3 that is ongoing and a bit delayed atm to add more digital twins for evaluation and/or training. The main use case for ManiSkill 3 and SIMPLER integration is faster evaluation as ManiSkill 3 generates visual data far faster in parallel compared to ManiSkill 2 / other simulators.

closed 2024-08-06 by yxchng