simpler-env/SimplerEnv #23: Does SimplerEnv support fine-tuning evaluation?

yxchng, 2024-08-06:
Many real-world evaluations of generalist robotic models require fine-tuning on data from target settings due to differences in action space, coordinate frames, and other contextual factors. Without such fine-tuning, models often fail when no data from similar settings are present. Evaluating models in this manner is crucial as it demonstrates the effectiveness of the pretrained knowledge they have acquired. Does SimplerEnv support fine-tuning on data from specific target settings for such evaluations?

xuanlinli17, 2024-08-07:
SIMPLER is agnostic to the model it evaluates (you can evaluate any policy you want to). Though, our current project is only about policy evaluation and not about policy finetuning.

yxchng, 2024-08-08:
How can I fine-tune for SimpleEnv? How to collect BC data from the simulation environment? Any guide?

xuanlinli17, 2024-08-08:
Simpler env serves as “test set” for robot manipulation  akin to supervised learning. You shouldn’t train on test set or the test set will be meaningless.

yxchng, 2024-08-08:
Hmmm, I am new to this field, but based on this https://github.com/openvla/openvla/issues/19 and this https://github.com/openvla/openvla/issues/17, model will not work out of the box if similar data are not part of the training mix. What am I missing here? 

OpenVLA seems to be performing badly on SimplerEnv right now.

xuanlinli17, 2024-08-08:
I think Google Robot and Bridge are in their training set, so you can evaluate OpenVLA immediately on Simpler (though I'm not sure if current OpenVLA's Bridge implementation is correct).

If you want to finetune OpenVLA, use rollouts from the Open-X-Embodiment dataset to train your model; don't use rollouts from Simpler as they are meant for "test set".

closed 2024-08-13 by xuanlinli17