simpler-env/SimplerEnv #24: [svulkan2] [error] GLFW error: X11: Failed to open display

armstronglzh, 2024-08-07:
hi,I met this error when I test :
python simpler_env/simple_inference_visual_matching_prepackaged_envs.py
Maybe I can't see the operation window because of this ,how can I solve this ,and also the warning:
[svulkan2] [warning] Continue without GLFW.

armstronglzh, 2024-08-07:
I solve this error to set  os.environ["DISPLAY"] = ":1" in the py file,but I still can't see the whole progress in a display window,is it normal or is there something else I don't set correctly.

xuanlinli17, 2024-08-07:
This is normal. You can ignore the warning.

armstronglzh, 2024-08-08:
> This is normal. You can ignore the warning.

I mean that after running the program, a window should appear to display the robot's running process,but I don't see any window.Is it normal or the program is just like this with no window displaying

xuanlinli17, 2024-08-08:
The script does not display a window. Instead, evaluation videos are saved. Please see the script for more details.

closed 2024-08-13 by xuanlinli17