simpler-env/SimplerEnv #25: How to adjust the camera angle in the output video to show more of the robotic arm?

wugh, 2024-08-12:
Hi,
I want to camera angle in the output video to show more of the robotic arm, but dont want to change the input of inference model, how to do it?

xuanlinli17, 2024-08-12:
For visual matching evaluation, unfortunately this is not possible to do (since we overlay the real background into simulated observations)

For variant aggregation evaluation, you can render images from a 3rd camera view. You can adjust the "human" camera config in the code.

wugh, 2024-08-12:
> For visual matching evaluation, unfortunately this is not possible to do (since we overlay the real background into simulated observations)
> 
> For variant aggregation evaluation, you can render images from a 3rd camera view. You can adjust the "human" camera config in the code.

thank you for your reply xuanlin, 

I think what you mean is script like `scripts/octo_pick_coke_can_variant_agg.sh` ?
I didn't find option to adjust the camera config, so this camera config is hard code in some file?

xuanlinli17, 2024-08-12:
Yes.

You can add cameras to the robot config file like https://github.com/simpler-env/ManiSkill2_real2sim/blob/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/mani_skill2_real2sim/agents/configs/google_robot/defaults.py

wugh, 2024-08-13:
> Yes.
> 
> You can add cameras to the robot config file like https://github.com/simpler-env/ManiSkill2_real2sim/blob/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/mani_skill2_real2sim/agents/configs/google_robot/defaults.py

ok, I will try this. thank you

closed 2024-09-03 by xuanlinli17