simpler-env/SimplerEnv #26: Installation

ddopplereffekt, 2024-08-12:
Hello,
The [full installation instructions](https://github.com/simpler-env/SimplerEnv#full-installation-rt-1-and-octo-inference-env-building) have the following lines:

"""
pip install tensorflow==2.15.0
pip install -r requirements_full_install.txt
pip install tensorflow[and-cuda]==2.15.1 # tensorflow gpu support
"""

I don't understand why first installing tensorflow 2.15.0 and then 2.15.1 with GPU support? Could someone please explain this, thank you!

xuanlinli17, 2024-08-13:
It's a hack to get around the weird tensorflow install dependencies; I don't remember directly installing tensorflow[and-cuda]==2.15.1 works.

closed 2024-09-03 by xuanlinli17

LukeLIN-web, 2025-03-11:
`dlimp 0.0.1 requires tensorflow==2.15.0, but you have tensorflow 2.15.1 which is incompatible.`