simpler-env/SimplerEnv #27: Issue with 'RuntimeWarning: Mean of empty slice.'

MedesL, 2024-08-13:
Hello guys, 

I encountered a strange problem when I tried to run **calc_metrics_evaluation_videos.py** to see the results of the evaluation. When I run this script I always get this message '_RuntimeWarning: Mean of empty slice._', followed by a lot of warning, like the one behind the arrow ==> **WARNING: avg_sim_success is nan** and balabala. 
My models are all downloaded followed by instructions and in the correct path, but I get these warnings regardless of which model, and the result is like 'vertical visual matching sim success {'rt-2-x': nan, 'rt-1-converged': nan, 'rt-1-15pct': nan, 'rt-1-x': nan, 'rt-1-begin': nan, 'octo-base': nan}', all empty value. 
Has anyone experienced a similar problem? If it's convenient, could you share your solution?

In the end, thanks for the amazing tools and the wisdom of the authors.

xuanlinli17, 2024-08-13:
I think it's because the videos aren't placed in the correct path? please check the script for more details.

MedesL, 2024-08-14:
> I think it's because the videos aren't placed in the correct path? please check the script for more details.

Hi Xuanlin,

Thank you very much for your answer, it makes me clear a lot! 
I now have the video generated by **simple_inference_visual_matching_prepackaged_envs.py** and correct path of the video. But **calc_metrics_evaluation_videos.py** still has the same problem, giving me a _WARNING: avg_sim_success is nan_ . 
I'm a little confused about what is the correct input for **calc_metrics_evaluation_video.py**?

xuanlinli17, 2024-08-14:
Oh, for `simple_inference_visual_matching_prepackaged_envs.py`, you can directly get the results from the reported success rates in the terminal. The `calc_metrics_evaluation_videos.py` is, on the other hand, used to reproduce the paper's evaluation results (i.e., correlation between sim & real using identical task, object pose, and robot pose setup) after you running all of `scripts/*` (or if you download all evaluation videos from huggingface and unzip them). It doesn't apply to the results of `simple_inference_visual_matching_prepackaged_envs.py`

MedesL, 2024-08-16:
Hi Xuanlin, 

Thank you very much for your clear explanation! 
And thank you for your patience and excellent work.

closed 2024-08-22 by MedesL