simpler-env/SimplerEnv #28: something about parameters

armstronglzh, 2024-08-14:
The command is like:
python simpler_env/main_inference.py --policy-model ${policy_model} --ckpt-path ${ckpt_path} \
  --robot ${robot} --policy-setup widowx_bridge \
  --control-freq 5 --sim-freq 500 --max-episode-steps 60 \
  --env-name PutCarrotOnPlateInScene-v0 --scene-name ${scene_name} \
  --rgb-overlay-path ${rgb_overlay_path} \
  --robot-init-x ${robot_init_x} ${robot_init_x} 1 --robot-init-y ${robot_init_y} ${robot_init_y} 1 --obj-variation-mode episode --obj-episode-range 0 24 \
  --robot-init-rot-quat-center 0 0 0 1 --robot-init-rot-rpy-range 0 0 1 0 0 1 0 0 1;

I want to know what does "robot-init-x,robot-init-y,obj-variation-mode,obj-episode-range,robot-init-rot-quat-center,robot-init-rot-rpy-range"    means

xuanlinli17, 2024-08-14:
`robot-init-x,robot-init-y,robot-init-rot-quat-center,robot-init-rot-rpy-range`

These are used to initiate a gridded sweep of robot poses.

`obj-variation-mode,obj-episode-range`

"obj-variation-mode" can be `xy` or `episode`; `xy` means sweeping evaluations using object xy locations (for single object task like pick coke can; not in the script you posted). `episode` means sweeping evaluations using episode ids, each episode id corresponds to some predefined object configurations (this applies when a task has >1 objects, e.g., move near)

armstronglzh, 2024-08-14:
> `robot-init-x,robot-init-y,robot-init-rot-quat-center,robot-init-rot-rpy-range`
>
> These are used to initiate a gridded sweep of robot poses.
>
> `obj-variation-mode,obj-episode-range` [...]

closed 2024-08-14 by armstronglzh