simpler-env/SimplerEnv #29: NameError: name 'OctoInference' is not defined.

armstronglzh, 2024-08-15:
I have download the octo files according to Octo Inference Setup,why still no octo model found

armstronglzh, 2024-08-15:
by the way I am running command:
python simpler_env/main_inference.py --policy-model octo-small --ckpt-path None --robot widowx --policy-setup widowx_bridge   --control-freq 5 --sim-freq 500 --max-episode-steps 60   --env-name StackGreenCubeOnYellowCubeBakedTexInScene-v0 --scene-name bridge_table_1_v1 --rgb-overlay-path ManiSkill2_real2sim/data/real_inpainting/bridge_real_eval_1.png  --robot-init-x 0.147 0.147 1 --robot-init-y 0.028 0.028 1 --obj-variation-mode episode --obj-episode-range 0 24   --robot-init-rot-quat-center 0 0 0 1 --robot-init-rot-rpy-range 0 0 1 0 0 1 0 0 1 --additional-env-save-tags octo_init_rng_0

armstronglzh, 2024-08-15:
And the folder is like:
SimplerEnv-main
     -octo
           -docs
           -examples
           -octo
           ....
     -simpler_env
           -main_inference.py
           ...
     -...

xuanlinli17, 2024-08-15:
I think you forgot to `cd octo/; pip install -e .`

armstronglzh, 2024-08-16:
> I think you forgot to `cd octo/; pip install -e .`
> 
> Please follow the instruciton in the Octo inference setup in README (it also requires installing jax)

I am sure I have done this all,the "pip install -e ." result is:
Successfully built octo
Installing collected packages: octo
Successfully installed octo-0.0.0

but still the error

V-Enzo, 2025-12-30:
It is because the jax version is too new. 

Installing the jax[cuda11_pip]==0.4.20 works.

pip install --upgrade "jax[cuda11_pip]==0.4.20" -f https://storage.googleapis.com/jax-releases/jax_cuda_releases.html