simpler-env/SimplerEnv #31: Any way to get expert demos for each task?

MasterXiong, 2024-08-19:
Hi, 

Thanks for open-sourcing this brilliant benchmark! I was wondering that is there any quick way to get expert demos for each task in Simpler? Like a script oracle policy, or a RL policy trained on each env?

I didn't find anything like these in the codebase. But I was wondering if the authors may have run some relevant experiments before and could help share some insights on this? Like how good can a RL policy perform in each env? Thanks!

xuanlinli17, 2024-08-19:
The "expert demos" are from the open-x-embodiment dataset.

MasterXiong, 2024-08-19:
Sorry for the confusion. I mean expert demos for each Simpler env in simulation. I guess directly running a open-x-embodiment action sequence in simpler would just fail due to dynamics mismatch?

xuanlinli17, 2024-08-19:
Open-loop unrolling demo trajectories will most likely succeed since simpler-env already performs sysid to match the dynamics gap between sim & real.

MasterXiong, 2024-09-03:
Thanks for your reply! If I want to run a OXE expert trajectory in a open-loop way in the corresponding `simpler` environment, how should I setup and reset the environment to match with those in the expert trajectory? Thanks!

xuanlinli17, 2024-09-03:
For existing environments in Simpler, you just need to take the trajectories with the same language instruction from the demo dataset and adjust the object pose to match the real eval

For environments not already present in this repo, you need to create new environment, which means you need the assets, a proxy simulation env, and a frame from demos to overlay the sim observation (with robot arm removed); you can generate assets via https://www.sudo.ai/

MasterXiong, 2024-09-04:
I see. Thanks! I wanted to run demos for existing `simpler` environments. I'll try with the first approach you suggest. Thanks!

MasterXiong, 2024-09-07:
Hi @xuanlinli17 , thanks a lot for your help!

About "adjust the initial object pose to match the demo's first frame" in your previous reply, I was wondering that how can I achieve this adjustment? I checked the OXE data and there seems to be no information about the demo's initial pose. Or should I do this manually by checking their visual appearance by myself? Could you help provide a simple example code of how to do the adjustment here? Thanks!

xuanlinli17, 2024-09-07:
You have to manually tune the object pose. You can use the nvdiffrast utilities in https://github.com/Jiayuan-Gu/GeTex (except baking texture).

closed 2024-09-22 by MasterXiong

IrvingF7, 2025-03-30:
Hi! @MasterXiong , I wonder if you ever succeeded at creating a dataset like what you described, and if so, could you share it?