simpler-env/SimplerEnv #32: Support for Droid dataset and evaluation

SiyuanHuang95, 2024-08-22:
Hi,

I am wondering whether you can integrate the robotic setup (Franka panda), and camera setup (2 overhead and 1 wrist camera) for the Droid dataset, since it is currently the largest one open-sourced robotic dataset?

xuanlinli17, 2024-08-23:
Unfortunately I don't have bandwidth for adding it; the visual matching setup also has the limitation of applying only to fixed camera setups (instead of dynamics camera setups like wrist cameras), though variant aggregation evaluations do not suffer from this limitation.

SiyuanHuang95, 2024-08-23:
Got It.

closed 2024-08-23 by SiyuanHuang95

LukeLIN-web, 2025-03-13:
> Unfortunately I don't have bandwidth for adding it; the visual matching setup also has the limitation of applying only to fixed camera setups (instead of dynamics camera setups like wrist cameras), though variant aggregation evaluations do not suffer from this limitation.

I wonder whether the  bridge env setup also has the limitation of applying only to fixed camera setups (instead of dynamics camera setups like wrist cameras) ?

xuanlinli17, 2025-03-13:
Yes, it's currently fixed camera setup.

LukeLIN-web, 2025-03-13:
> Yes, it's currently fixed camera setup.

Thank you. What is the reason? Is it because physical setting or software? 
Is it possible for me to open a PR to add **wrist cameras** in bridge ?  If it is possible, how many coding time to may need.

xuanlinli17, 2025-03-13:
The wrist camera moves which breaks visual matching setup. The variant aggregation setup is fine w/ wrist camera.
You can modify the robot agent. It should be fast.