simpler-env/SimplerEnv #34: RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed

hahans, 2024-08-24:
I run  python simpler_env/simple_inference_visual_matching_prepackaged_envs.py --policy openvla/openvla-7b \
        --ckpt-path None --task google_robot_move_near_v1  --logging-root ./results_simple_eval/  --n-trajs 10
I checked the troubleshooting section and followed the link's instructions,but it still didn't work.
Meet the error : RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed
GPU : RTX3090
Ubuntu22.04

hahans, 2024-08-24:
I added a josn file, this error is gone, but I do not see the robot simulation screen

xuanlinli17, 2024-08-24:
No screen should appear. Evaluation videos should be automatically recorded. You can find them at `results/`

closed 2024-09-03 by xuanlinli17

chinhsuanwu, 2025-04-29:
Solved it by following this [doc](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html)