simpler-env/SimplerEnv #35: Migrate example environments to ManiSkill 3

StoneT2000, 2024-08-29:
Intention is to replace the currently git submoduled enviroments with the new GPU accelerated ManiSkill 3 package and some better packaged code designed to be easily extended.

StoneT2000, 2025-04-02:
See branch https://github.com/simpler-env/SimplerEnv/tree/maniskill3 for how to evaluate some baselines on the gpu parallelized rendering envs. We only support 4 of the envs at the moment on the GPU version.

sattyamjjain, 2026-09-02:
Is this still wanted? Happy to take a piece of it.

I run evaluation against ManiSkill-side environments already, so the migration path is something I have to understand anyway. If the blocker is that nobody has time rather than that the design is unsettled, say so, and I will open a scoped PR for one environment first rather than a big one.