simpler-env/SimplerEnv #37: How to run the environment on CPU

MasterXiong, 2024-09-07:
Hi, 

I was wondering that how can I run the environment on CPU? As the default config seems to be running on GPU, but I found it hard to parallelize the environments with GPU ([link](https://github.com/simpler-env/SimplerEnv/issues/36#issue-2502007201)). Not sure if this issue can be solved by simulating on CPU instead? Thanks!

xuanlinli17, 2024-09-07:
Simpler is based on ManiSkill2 and rendering requires a GPU. 
However, ManiSkill3 will support both GPU parallelization and CPU rendering. @StoneT2000  is currently migrating the environments.

MasterXiong, 2024-09-08:
Thanks! Just to confirm, so there is no way to run multiple environments with the current version based on ManiSkill2 right?

xuanlinli17, 2024-09-08:
You can run multiple envs, but it's a bit tricky; however you can always split your eval script into multiple scripts (but you need to load the networks multiple times in this case).

MasterXiong, 2024-09-08:
Thanks! Can you help give some more details on how to implement this tricky way to run multiple envs? I'm trying to accelerate my evaluation pipeline by feeding a large batch of observations to a model insteading of processing only one env observation at a time.

xuanlinli17, 2024-09-08:
run many different bash scripts, each like seed 0-1000, seed 1000-2000, etc

closed 2024-09-08 by MasterXiong