simpler-env/SimplerEnv #39: How to specify which objects to manipulate in pick and move tasks

MasterXiong, 2024-09-22:
Hi, 

I was wondering how can I specify which object(s) to manipulate in the `pick objects` and `move A to B` tasks? Based on my understanding of the code, the candidate set of possible objects are defined like [here](https://github.com/simpler-env/ManiSkill2_real2sim/blob/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/mani_skill2_real2sim/envs/custom_scenes/move_near_in_scene.py#L611). Then I should change the `options` input to the `reset` function to specify which object(s) to use in the environment. Is my understanding correct? Thanks!

xuanlinli17, 2024-09-23:
The two envs have different initialization mechanisms. `move A to B` uses episode-based randomization, so you can do so by passing in `episode_id` like https://github.com/simpler-env/SimplerEnv/blob/0b13b698138576c449f8853c99b04202e3c04952/simpler_env/evaluation/maniskill2_evaluator.py#L82 . For `pick objects` , you can do so by passing in different seeds in the options

MasterXiong, 2024-09-24:
Thanks a lot! A further question is that it seems that only a subset of objects or triplets of objects are used for the pick and move tasks (like [here](https://github.com/simpler-env/ManiSkill2_real2sim/blob/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/mani_skill2_real2sim/envs/custom_scenes/grasp_single_in_scene.py#L594) and [here](https://github.com/simpler-env/ManiSkill2_real2sim/blob/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/mani_skill2_real2sim/envs/custom_scenes/move_near_in_scene.py#L613)). I was wondering that how are these task configurations chosen? Are they chosen to reflect the objects that are seen in the OXE demos? If that's the case, can I add more configurations to test out-of-distribution generalization to unseen objects or triplets in OXE? Thanks!

xuanlinli17, 2024-09-24:
These are not necessarily seen by OXE demos; they are just randomly chosen.

MasterXiong, 2024-09-24:
Where can I find a full list of all the possible objects supported in `simpler` for picking and moving tasks then?

xuanlinli17, 2024-09-24:
Checkout these json files at https://github.com/simpler-env/ManiSkill2_real2sim/tree/cd45dd27dc6bb26d048cb6570cdab4e3f935cc37/data/custom which are loaded in the envs

closed 2024-11-14 by xuanlinli17