simpler-env/SimplerEnv #40: How to convert RT-1-X action to Octo action

MasterXiong, 2024-09-24:
Hi, 

I wanted to use successful episodes collected by RT-1-X in the `simpler` env as demos to fine-tune an Octo model. I was wondering that how could I convert the raw action generated by RT-1-X to the corresponding raw action of Octo? As I note that the inference function for the two models are different in `simpler` code, which may indicate that they do not use the same action space and some coordinates transformation is required here? Thanks!

xuanlinli17, 2024-09-24:
Just compare the 2 inference code. I think only the gripper is different.

MasterXiong, 2024-09-25:
Thanks! But I noticed that for the actions dimensions of orientations, RT-1 uses axis angle representation for google robot tasks, and roll, pitch, yaw representation for widowx tasks ([link](https://github.com/simpler-env/SimplerEnv/blob/0b13b698138576c449f8853c99b04202e3c04952/simpler_env/policies/rt1/rt1_model.py#L45)). And Octo seems to use roll, pitch, yaw for both kinds of robots. So should I transform axangle to euler for google robot?

xuanlinli17, 2024-09-25:
I think so.

closed 2024-11-14 by xuanlinli17