simpler-env/SimplerEnv #44: AttributeError: 'TimeLimit' object has no attribute 'get_language_instruction'

branyang02, 2024-10-17:
https://colab.research.google.com/github/simpler-env/SimplerEnv/blob/main/example.ipynb no longer works.

The error is:
AttributeError: 'TimeLimit' object has no attribute 'get_language_instruction'

[image]

younik, 2024-10-18:
@branyang02 This is due to Gymnasium 1.0; use `env.unwrapped.get_language_instruction()` instead, or downgrade Gymnasium.

xuanlinli17, 2024-10-19:
Forced gymnasium < 1.0 in latest commit

closed 2024-11-14 by xuanlinli17

LukeLIN-web, 2025-04-07:
Just for a note for others like me . https://github.com/simpler-env/SimplerEnv/commit/f3035e0135d102587f2acd25e5cada3189c06d0b