simpler-env/SimplerEnv #47: In multiple GPUs: RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed #20

DelinQu, 2024-10-31:
Hi 👋, I caught the error "RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed #20" when starting the environment in a non-zero device. The simulator works well when using the default CUDA device 0, but it failed in any other devices in `1,2,3,4,5,6,7`:

I tried the solution at https://github.com/simpler-env/SimplerEnv/issues/20, but it doesn't work.
[image]

xuanlinli17, 2024-10-31:
You can try ManiSkill3 version of the Bridge envs for multi-GPU evaluation. @StoneT2000 will migrate the Google Robot envs later.

xuanlinli17, 2024-10-31:
See e.g.., https://github.com/haosulab/ManiSkill/issues/79

StoneT2000, 2024-10-31:
for maniskill 3 the fix should work proposed by xuanlin should work

DelinQu, 2024-11-01:
Thanks for your replies, the DISPLAY has already been unset in https://github.com/simpler-env/SimplerEnv/blob/d55e19162be86794875839725fd484b768e25873/simpler_env/main_inference.py#L21C2-L21C31, I have no idea why it doesn't work for me. So I will migrate the environment to maniskiill3, does it cause any difference to the evaluation results, compared with mainiskill2? I must make the evaluation fair.

xuanlinli17, 2024-11-01:
It should be very similar.

DelinQu, 2024-11-01:
It's wired that simpler start-up successfully if I set the `CUDA_VISIBLE_DEVICES=x,0 python {}`. The Memory and utils of `CUDA:0` are almost zero, but it's critical for setup maniskill2 environments:

[image]

DelinQu, 2025-02-10:
> You can try ManiSkill3 version of the Bridge envs. [@StoneT2000](https://github.com/StoneT2000) will migrate the Google Robot envs later.
> 
> I think a fix for ManiSkill2 is to set `DISPLAY="" CUDA_VISIBLE_DEVICES=x python {}`

Hi, thanks so much for your great efforts on this project. Could you kindly let me know when the Google Robot migration is expected to be completed?

StoneT2000, 2025-02-12:
Might be a while sorry. I can try taking a look again next week perhaps but my initial avoidance was trying to avoid diving too deep into the ruckig package.

Alternatively maybe i can setup one of our existing controllers and if the sr and mmrv values are good then maybe we can skip ruckig despite not modeling the real world controller perfectly

closed 2025-03-12 by xuanlinli17