simpler-env/SimplerEnv #50: Visual Texture Randomization for Maniskill3 branch

AlexandreBrown, 2024-11-06:
Hi,  
I would like to know how to alter the visual configuration of the env/task (eg: randomize texture between episodes, randomize lighting, camera fov etc).  

The documentation provided [here](https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/advanced.html#texture-modification) does not show an example.   
  
Any pointer is appreciated!

AlexandreBrown, 2024-11-06:
I managed to change the `base_color` but I cannot randomize or change the texture, any pointer is appreciated.   
```python
base_color_texture = RenderTexture2D(
    "/home/user/Downloads/cliff_side_4k.blend/textures/cliff_side_diff_4k.jpg"
)
for actor_name in self.base_env.unwrapped.scene.actors.keys():
    for part in self.base_env.unwrapped.scene.actors[actor_name]._objs:
        for triangle in (
            part.find_component_by_type(sapien.render.RenderBodyComponent)
            .render_shapes[0]
            .parts
        ):
            # triangle.material.set_base_color([0.8, 0.1, 0.1, 1.0])
            triangle.material.set_base_color_texture(base_color_texture)

obs_dict, _ = self.base_env.reset()
```

xuanlinli17, 2024-11-06:
@StoneT2000

StoneT2000, 2024-11-07:
Hi @AlexandreBrown. I will address the issue on the ManiSkill repo that you raised, thanks for raising it.

closed 2024-11-07 by StoneT2000