simpler-env/SimplerEnv #54: New environment

abadithela, 2024-12-07:
Hi,

This might be a simple fix but so far I am having trouble solving it. I am instantiating a new custom environment inside mani_skill2_real2sim/envs/custom_scenes, and then trying to test it out in the demo_manual_control_custom_envs.py. The issue is the environment ID is not getting registered by gym. Do I need to explicitly import the new custom environment somewhere? The only interface I see is in the __init__.py inside the envs folder, which has: 

`from .custom_scenes import *`

Here is the error I am seeing when I try to run the demo command:

```
python mani_skill2_real2sim/examples/demo_manual_control_custom_envs.py -e PutCarrotOnPlateInScene-v0_IROM --enable-sapien-viewer     -c arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos -o rgbd robot widowx sim_freq @500 control_freq @5     scene_name bridge_table_1_v1  rgb_overlay_mode debug rgb_overlay_path data/real_inpainting/bridge_real_eval_1.png rgb_overlay_cameras 3rd_view_camera
```
Full Error:

```
opts: ['robot', 'irom_widowx', 'sim_freq', '@500', 'control_freq', '@5', 'scene_name', 'bridge_table_1_v1', 'rgb_overlay_mode', 'debug', 'rgb_overlay_path', 'data/real_inpainting/bridge_real_eval_1.png', 'rgb_overlay_cameras', '3rd_view_camera']
env_kwargs: {'robot': 'irom_widowx', 'sim_freq': 500, 'control_freq': 5, 'scene_name': 'bridge_table_1_v1', 'rgb_overlay_mode': 'debug', 'rgb_overlay_path': 'data/real_inpainting/bridge_real_eval_1.png', 'rgb_overlay_cameras': '3rd_view_camera'}
Traceback (most recent call last):
  File "/home/apurva/software/RapidEvalPPI/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/examples/demo_manual_control_custom_envs.py", line 504, in <module>
    main()
  File "/home/apurva/software/RapidEvalPPI/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/examples/demo_manual_control_custom_envs.py", line 153, in main
    env: BaseEnv = gym.make(
  File "/home/apurva/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 741, in make
    env_spec = _find_spec(id)
  File "/home/apurva/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 527, in _find_spec
    _check_version_exists(ns, name, version)
  File "/home/apurva/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 393, in _check_version_exists
    _check_name_exists(ns, name)
  File "/home/apurva/miniconda3/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 370, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `PutCarrotOnPlateInScene-v0_IROM` doesn't exist. Did you mean: `PutCarrotOnPlateInScene`?
```
Additional information: The environment is registered in the script mani_skill2_real2sim/envs/custom_scenes/irom_put_carrot_on_plate.py as follows:

```
# Initializes scene for carrot on plate task
@register_env("PutCarrotOnPlateInScene-v0_IROM", max_episode_steps=60)
class PutCarrotOnPlateInSceneIROM(PutOnBridgeInSceneEnvIROM):
    def __init__(self, **kwargs):
        source_obj_name = "bridge_carrot_generated_modified"
        target_obj_name = "bridge_plate_objaverse_larger"

        xy_center = np.array([-0.16, 0.00])
        half_edge_length_x = 0.075
        half_edge_length_y = 0.075
        grid_pos = np.array([[0, 0], [0, 1], [1, 0], [1, 1]]) * 2 - 1
        grid_pos = (
            grid_pos * np.array([half_edge_length_x, half_edge_length_y])[None]
            + xy_center[None]
        )

        xy_configs = []
        for i, grid_pos_1 in enumerate(grid_pos):
            for j, grid_pos_2 in enumerate(grid_pos):
                if i != j:
                    xy_configs.append(np.array([grid_pos_1, grid_pos_2]))

        quat_configs = [
            np.array([euler2quat(0, 0, np.pi), [1, 0, 0, 0]]),
            np.array([euler2quat(0, 0, -np.pi / 2), [1, 0, 0, 0]]),
        ]

        super().__init__(
            source_obj_name=source_obj_name,
            target_obj_name=target_obj_name,
            xy_configs=xy_configs,
            quat_configs=quat_configs,
            **kwargs,
        )

    def get_language_instruction(self, **kwargs):
        return "put carrot on plate"
```

xuanlinli17, 2024-12-08:
https://github.com/simpler-env/ManiSkill2_real2sim/blob/ef7a4d4fdf4b69f2c2154db5b15b9ac8dfe10682/mani_skill2_real2sim/envs/custom_scenes/__init__.py

Have you imported the environment here?

abadithela, 2024-12-09:
Thanks; I missed this. It is fixed now.

closed 2024-12-09 by abadithela