simpler-env/SimplerEnv #57: ImportError: cannot import name 'sa' from 'simulated_annealing'

revquiet, 2024-12-13:
I encountered an issue when trying to import sa from the simulated_annealing module in `tools/sysid/sysid.py.` The import statement fails so i  installed the module using pip:
```
pip install simulated-annealing
```
the following error arises:
```
cannot import name 'sa' from 'simulated_annealing' (/home/test/miniconda3/envs/simpler_env/lib/python3.10/site-packages/simulated_annealing/__init__.py) 
  File "/home/test/Code/git_repo/SimplerEnv/tools/sysid/sysid.py", line 11, in <module>
    from simulated_annealing import sa
ImportError: cannot import name 'sa' from 'simulated_annealing' (/home/test/miniconda3/envs/simpler_env/lib/python3.10/site-packages/simulated_annealing/__init__.py)
```
and the version of the simulated-annealing is 0.4.0. I checked the __init__.py file in the installed module directory and found no reference to sa. If sa is not available, it would be helpful to provide an example in the documentation of the correct usage.

revquiet, 2024-12-13:
When I modified the code as follows:
```
import simulated_annealing as sa
```

I encountered an error indicating that the sa module does not have the minimize() method. How can I correctly import this module and perform the corresponding system identification?
```
opt = sa.minimize(
    opt_fxn,
    init_action,
    opt_mode="continuous",
    step_max=2000,
    t_max=1.5,
    t_min=0,
    bounds=[[0, 1]] * (len(init_stiffness) * 2),
)
```

Could you please advise on the appropriate steps to resolve this issue?

xuanlinli17, 2024-12-13:
Did you `pip install git+https://github.com/nathanrooy/simulated-annealing`?

revquiet, 2024-12-16:
Thank you for your response. Your solution indeed resolved my issue. During the process of identifying the system, I used my own dataset instead of utilizing RT-1 and OCTO, thus skipping some crucial related steps.

closed 2024-12-16 by revquiet