simpler-env/SimplerEnv #58: Object placement in the scene

abadithela, 2024-12-14:
Hi,

I'm having an issue with placing an object correctly onto the table surface. The object seems to "sink" into the surface after it settles (see image below). It could be that while developing code, I might have changed code that is now resulting in this. Where does ManiSkill decide the "surface" of the table to be? I assumed that this would be encoded in the .glb file. I even downloaded a fresh version of the bridge_table_1_v1.glb to rule out the possibility that the mistake is in the .glb file. However, this issue still persists.

Thanks,
Apurva

[image]

StoneT2000, 2024-12-16:
This means the object is not modelled perfectly probably. Is this your own object or is this the original one from this repo? And are you using the batched version (maniskill3) of simpler or the original cpu based one (maniskill2)

abadithela, 2025-01-02:
I was using ManiSkill2. The issue was with the scene_pose; I accidentally modfified the default setting in base_env.py. Is Simpler now compatible with ManiSkill3?

StoneT2000, 2025-01-05:
4 of Simpler environments (bridge dataset) are officially supported in the faster ManiSkill3. See the ManiSkill3 branch: https://github.com/simpler-env/SimplerEnv/tree/maniskill3

It is eventually planned to support the other 4 environments (using google robot) but the timeline for that is uncertain at the moment.

closed 2025-03-12 by xuanlinli17