simpler-env/SimplerEnv #59: Google robot end-effector pose

allenzren, 2024-12-28:
Hi! Not sure if this is the right place to ask this, but do you happen to know, which frame of the Google Robot is the EE pose data collected in the fractal20220817_data dataset w.r.t?

I see [here](https://github.com/simpler-env/ManiSkill2_real2sim/blob/87dc84508520310e61c972ece399a0f034095e42/mani_skill2_real2sim/agents/configs/google_robot/defaults.py#L141-L142) the end-effector is defined as `link_gripper_tcp`, but I am not sure if the fractal dataset treats that as the end-effector.

Thanks!

xuanlinli17, 2024-12-29:
The simpler policy output EE should match the fractal EE. Specifically, next_rot = action_rot * cur_rot (wrt base frame), next_transl = action_transl + cur_transl (wrt base frame.

allenzren, 2024-12-29:
I see, thanks very much Simon! The ee-align part makes sense.

My question was actually more about the **proprioception** EE pose saved in the fractal data. But I suppose, since Simpler uses delta EE including the rotation for the controller, and rt1 works in Simpler, the EE frame defined for action in fractal should be the same as Simpler, and presumably that EE frame for action in fractal is the same as that for proprioception.

allenzren, 2024-12-29:
Oh actually this is not true any more with "ee-aligning"? suppose there is an actual EE frame in fractal that has a fixed translation from `link_gripper_tcp`, and ee_aligning still works. But the propriocetion may have a fixed translation offset now between fractal and simpler

closed 2024-12-29 by allenzren

xuanlinli17, 2024-12-29:
"ee-align" is just a name for decoupling rotation and translation actions (next_rot = action_rot * cur_rot (wrt base frame), next_transl = action_transl + cur_transl (wrt base frame)), instead of multiplying 2 SE(3) matrices (in which case new_pose = action_se3 @ cur_pose_se3). "ee-align" doesn't change the location of ee frame. `link_gripper_tcp` is set exactly at the fractal dataset's ee frame (since it is defined in urdf)

allenzren, 2024-12-29:
Got it! If `link_gripper_tcp` is set exactly at the fractal dataset's ee frame, then there should be no issue. Thanks for the detailed explanation!