simpler-env/SimplerEnv #60: System Identification

offjangir, 2025-01-01:
I have been trying to run the system identification scripts that have been provided. They do run good but simulated annealing takes too much time for me to complete. Currently, the code is set for 2000 iterations. Did you run it for 2000 iterations? how much time did it take to complete? what were the host system specifications that you ran it on.

xuanlinli17, 2025-01-02:
I typically run it for 400 iterations and initialize 3 rounds of sysid using the best result from the prior iteration. 2000 is just for illustrative purposes in case you want to let it run overnight and wait till the next day to see the result

offjangir, 2025-01-02:
Hi @xuanlinli17  I am running the direct example code. The code takes about 6 hours just for a 200 Iterations. I have been trying to do this multiple times.  I am running this on a 48 core AMD processor with 80GB A100. Are you sure this is the latest code you are using ?

xuanlinli17, 2025-01-03:
This is quite strange since multiprocessing is already automatically handled; but since you have A100 you can def use larger pool than 18: https://github.com/simpler-env/SimplerEnv/blob/main/tools/sysid/sysid.py#L151

offjangir, 2025-01-08:
I think this is a vulkan issue ? Maybe my processing is defaulting to CPU?

xuanlinli17, 2025-01-08:
Is your gpu utilization >0? you can try "time.time()" to benchmark time elapsed for stepping the environment. The simulated annealing is on cpu though.

abadithela, 2025-03-10:
Hi @xuanlinli17, the default WidowX configuration that you have released with Simpler is after performing sys ID correct?

xuanlinli17, 2025-03-10:
> Hi [@xuanlinli17](https://github.com/xuanlinli17), the default WidowX configuration that you have released with Simpler is after performing sys ID correct?

Yes

closed 2026-10-06 by offjangir