simpler-env/SimplerEnv #61: maniskill3 environment problem

Lucky-Light-Sun, 2025-01-05:
Hello, I'm new to the robotics domain and have explored SimplerEnv. I find it is an really interesting and  impressive work for robotic development! 

Last month, I used SimplerEnv with maniskill2 version. But I found it's too slow to evaluate octo/openvla models. So I want to switch to maniskill3 version to accelerate evaluation process. However when I install this new version, some problems happened.

***First***, there maybe some mistakes in you maniskill3 readme file.
[image]

***Second***, when I followed installation and Full Installation part. There are still some errors.
To be specific, when I run `bash scripts/octo_bridge.sh`, `ModuleNotFoundError` happened.

```
  File "/mnt/hdd1/xingyouguang/projects/robotics/simpler-env3/simpler_env/main_inference.py", line 6, in <module>
    from simpler_env.evaluation.argparse import get_args
  File "/mnt/hdd1/xingyouguang/projects/robotics/simpler-env3/simpler_env/__init__.py", line 2, in <module>
    import mani_skill2_real2sim.envs
ModuleNotFoundError: No module named 'mani_skill2_real2sim'
```
I think, there are some reasons caused this problem:

1. I just run `git clone https://github.com/simpler-env/SimplerEnv`, but also need to git checkout some version. So i git checkout maniskill3. However, I do not find any info about this maniskill3 version in ReadMe file.
2. I switch to maniskill3, but why this error report mani_skill2 ??? I can run `.gitmodules` to clone https://github.com/simpler-env/ManiSkill2_real2sim. But i think this is not correct, because this may convert to maniskill2 environment.
3. I checked `git+https://github.com/haosulab/ManiSkill.git` rep. However, I want to convert it to maniskill2_sim2real. But I do not see their corelations. Because the overlay file does not exists in ManiSkill3 rep.

Could you please help me to solve these problems? Thank you again for your contributions to the community!

Looking forward to your reply!!

xuanlinli17, 2025-01-06:
@StoneT2000

Lucky-Light-Sun, 2025-01-13:
fixed

closed 2025-01-13 by Lucky-Light-Sun

IrvingF7, 2025-01-16:
Hi @Lucky-Light-Sun. Could you talk about how to solve this? Really appreciate it!

Lucky-Light-Sun, 2025-02-06:
> Hi [@Lucky-Light-Sun](https://github.com/Lucky-Light-Sun). Could you talk about how to solve this? Really appreciate it!

Checkout the branch, and then build the environment as README. 
The most important thing is that `scripts/*.sh` should not be used any more. We should `simpler_env/real2sim_eval_maniskill3.py` as mentioned in README.

Akila-Ayanthi, 2025-02-07:
@Lucky-Light-Sun were you able to setup openvla with maniskill3?

Lucky-Light-Sun, 2025-02-08:
> [@Lucky-Light-Sun](https://github.com/Lucky-Light-Sun) were you able to setup openvla with maniskill3?
I didn't have time to modify it last month. But I will do this in the next two weeks.

Akila-Ayanthi, 2025-02-19:
@Lucky-Light-Sun Any updates?