simpler-env/SimplerEnv #64: About [svulkan2] [error]  OptiX Error

leonwu117, 2025-02-11:
I have a problem, except bridge environment, other environments will report errors. Does anyone know how to fix it?

It show:
 [2025-02-11 20:26:48.931] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INTERNAL_ERROR
[2025-02-11 20:26:48.931] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INTERNAL_ERROR
[2025-02-11 20:26:48.931] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INTERNAL_ERROR
[2025-02-11 20:26:49.435] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.435] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.441] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.453] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.453] [svulkan2] [error]  OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.458] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.498] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.505] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.591] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.597] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.636] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
[2025-02-11 20:26:49.642] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE

jasper0314-huang, 2025-02-16:
Hi @leonwu117,

I encountered the same issue. Have you found a solution?

jasper0314-huang, 2025-02-19:
I found the solution here: https://github.com/mitsuba-renderer/mitsuba3/issues/1019#issuecomment-1864827711
Specifically, I copied the `/usr/share/nvidia` folder from the local server into the Docker container, and it works now.

leonwu117, 2025-03-06:
> I found the solution here: [mitsuba-renderer/mitsuba3#1019 (comment)](https://github.com/mitsuba-renderer/mitsuba3/issues/1019#issuecomment-1864827711) Specifically, I copied the `/usr/share/nvidia` folder from the local server into the Docker container, and it works now.

Thank you for response!  But I'm using a remote server(SSH), not docker. Now I wonder if the lack of monitors is the reason

jasper0314-huang, 2025-03-06:
> Thank you for response! But I'm using a remote server(SSH), not docker. Now I wonder if the lack of monitors is the reason

Hi @leonwu117 ,
I also use the SSH and run the docker image in remove server without monitors. The image is built in local with `/usr/share/nvidia` from local, too.

leonwu117, 2025-03-06:
> Hi [@leonwu117](https://github.com/leonwu117) ,  你好， I also use the SSH and run the docker image in remove server without monitors. The image is built in local with `/usr/share/nvidia` from local, too.我也使用 SSH 并在没有显示器的远程服务器上运行 Docker 镜像。镜像也是在本地使用 `/usr/share/nvidia` 从本地构建的。

Appreciate that! I would try to build docker first

leonwu117, 2025-03-08:
1. Check if nvoptix.bin Already Exists： `sudo find / -name nvoptix.bin 2>/dev/null` , if not, just Verify NVIDIA Driver Version： `nvidia-smi`
2.  Download the Corresponding NVIDIA Driver： [](https://www.nvidia.com/download/index.aspx)
3. 
```
chmod +x NVIDIA-Linux-x86_64-550.107.02.run
./NVIDIA-Linux-x86_64-550.107.02.run --extract-only
```
4. `sudo cp NVIDIA-Linux-x86_64-550.107.02/nvoptix.bin /usr/share/nvidia/`

closed 2025-03-10 by leonwu117