simpler-env/SimplerEnv #68: RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent

Jianing2002, 2025-02-18:
Hello,
﻿
I encountered an error while running the code from your repository. The error message is as follows:
﻿
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent
﻿
The error occurs after running 10 episodes on all policies. The error is triggered at the following line in the code:
﻿
self._renderer = sapien.SapienRenderer(**renderer_kwargs)
﻿
I have checked the following:
GPU driver is up-to-date.
Vulkan configuration files are present.
All dependencies are correctly installed.
﻿
However, the issue persists. Based on similar issues reported in other projects, it seems this might be related to the Vulkan renderer configuration.
Could you please provide some guidance on how to resolve this error? Any help would be greatly appreciated.
Thank you!

xuanlinli17, 2025-02-19:
Try `DISPLAY="" CUDA_VISIBLE_DEVICES=0 python {script}`?

Jianing2002, 2025-02-20:
> Try `DISPLAY="" CUDA_VISIBLE_DEVICES=0 python {script}`?

Thank you for your reply. Actually, I am using CUDA0 and still have this issue

Liang-ZX, 2025-02-25:
> Try `DISPLAY="" CUDA_VISIBLE_DEVICES=0 python {script}`?

CUDA_VISIBLE_DEVICES=0 works for me. But why other gpus cannot work?

xuanlinli17, 2025-03-10:
If you have DISPLAY="" all gpus should work? It's a legacy issue of sapien2.x

KevinGuo07, 2025-03-12:
Same problem here, I checked the vulkan and nvidia driver, both are ok, after setting 
"
export VK_ICD_FILENAMES=""
export VK_LAYER_PATH=""
export MESA_NO_ACCEL=1
export SVULKAN2_DISABLE=1
export DISPLAY=""
export SAPIEN_RENDERER=cpu
export PYOPENGL_PLATFORM=egl
"
this problem still appears and seems to try display with renderer, how to figure this out?

LukeLIN-web, 2025-04-11:
I also meet this `RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent` after eval, but looks fine. don't affect eval

LukeLIN-web, 2025-04-26:
> If you have DISPLAY="" all gpus should work? It's a legacy issue of sapien2.x

DISPLAY="" is not work for me . I will try CUDA 0

chinhsuanwu, 2025-04-29:
Solved it by following this [doc](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html)

LukeLIN-web, 2025-04-30:
> Solved it by following this [doc](https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html)

Thank you , could you point out which commands are the solution?

chinhsuanwu, 2025-04-30:
use `vulkan-tools` instead

Boltzmachine, 2025-06-04:
This problem randomly occurs to me. Sometime it works sometimes it does not.

Taokt, 2025-06-05:
> Hello, ﻿ I encountered an error while running the code from your repository. The error message is as follows: ﻿ RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent ﻿ The [...]

Check if you have libegl1 install, based on [this post](https://github.com/NVIDIA/nvidia-container-toolkit/issues/191#issuecomment-2022154630), rendering is actually supported by  EGL, not GLX. For me I was having this error in Docker container, and I solved it by 'apt install libegl1'.

QQQQQQ11234, 2025-11-29:
> This error does not appear on other environments but appears only on `move_near_variant_agg`. And for this env, it occurs only after some episodes.

How do you solve it?

QQQQQQ11234, 2025-11-29:
> This error does not appear on other environments but appears only on `move_near_variant_agg`. And for this env, it occurs only after some episodes.

DISPLAY="" CUDA_VISIBLE_DEVICES=0 VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia_icd.json    
I solved it by this