simpler-env/SimplerEnv #69: Segmentation Error when existing program

chang-github-00, 2025-02-18:
**If an env is created and then the program exits, there would be a segmentation error (signal `SIGSEGV`)**

Checked vulkaninfo seems correct. 
```Layers: count = 2
=================
VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 9
		GPU id = 0 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 1 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 2 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 3 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 4 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 5 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 6 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 7 (NVIDIA A800 80GB PCIe)
		Layer-Device Extensions: count = 0

		GPU id = 8 (llvmpipe (LLVM 18.1.8, 256 bits))
		Layer-Device Extensions: count = 0
```
vulkaninfo | head -n 5
```

'DISPLAY' environment variable not set... skipping surface info
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.283
```
Installed following readme instructions, and set numpy version to 1.26.4 

When testing the `example.py`

```
import simpler_env
from simpler_env.utils.env.observation_utils import get_image_from_maniskill2_obs_dict

import threading

lock = threading.Lock()

env = simpler_env.make('google_robot_pick_coke_can')
obs, reset_info = env.reset()
instruction = env.get_language_instruction()
print("Reset info", reset_info)
print("Instruction", instruction)

done, truncated = False, False
id = 0
while not (done or truncated) and id<10:
   # action[:3]: delta xyz; action[3:6]: delta rotation in axis-angle representation;
   # action[6:7]: gripper (the meaning of open / close depends on robot URDF)
   image = get_image_from_maniskill2_obs_dict(env, obs)
   action = env.action_space.sample() # replace this with your policy inference
   
   obs, reward, done, truncated, info = env.step(action) # for long horizon tasks, you can call env.advance_to_next_subtask() to advance to the next subtask; the environment might also autoadvance if env._elapsed_steps is larger than a threshold

   new_instruction = env.get_language_instruction()
   id += 1
   if new_instruction != instruction:
      # for long horizon tasks, we get a new instruction when robot proceeds to the next subtask
      instruction = new_instruction
      print("New Instruction", instruction)
   
episode_stats = info.get('episode_stats', {})
print("Episode stats", episode_stats)
```

The program could be successfully runned, yet have segmentation error at the end.

```
[2025-02-18 16:43:45.649] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2025-02-18 16:43:45.649] [svulkan2] [warning] Continue without GLFW.
/home/data/anaconda3/envs/lapa/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.get_language_instruction to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.get_language_instruction` for environment variables or `env.get_wrapper_attr('get_language_instruction')` that will search the reminding wrappers.
  logger.warn(
Reset info {'scene_name': 'google_pick_coke_can_1_v4', 'scene_offset': None, 'scene_pose': None, 'scene_table_height': 0.87, 'urdf_version': 'recolor_tabletop_visual_matching_1', 'rgb_overlay_path': '/ssddata/junxianh/Agent-Projects/Latent-Planning/SimplerEnv/ManiSkill2_real2sim/data/real_inpainting/google_coke_can_real_eval_1.png', 'rgb_overlay_cameras': ['overhead_camera'], 'rgb_overlay_mode': 'background', 'disable_bad_material': False, 'model_id': 'opened_coke_can', 'model_scale': 1.0, 'distractor_model_ids': None, 'distractor_model_scales': None, 'obj_init_pose_wrt_robot_base': Pose([0.587925, -0.0238302, 0.840576], [0.707052, -0.0081018, -0.01162, -0.70702]), 'orientation': 'laid_vertically'}
Instruction pick coke can
/home/data/junxianh/anaconda3/envs/lapa/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.robot_uid to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.robot_uid` for environment variables or `env.get_wrapper_attr('robot_uid')` that will search the reminding wrappers.
  logger.warn(
Episode stats OrderedDict([('n_lift_significant', 0), ('consec_grasp', False), ('grasped', False)])
Segmentation fault (core dumped)
```

chang-github-00, 2025-02-18:
I tried upgrading gymnasium to 1.0.0,

closing the code with only `env = simpler_env.make('google_robot_pick_coke_can')` no longer causes segmentation fault, but leads to a series of incompatible problems.

so possibly related to gymnasium version.

xuanlinli17, 2025-02-19:
You can ignore the segfault since it's due to sapien2's problem.

closed 2025-02-20 by chang-github-00