simpler-env/SimplerEnv #71: Failed to build maniskill2_environment

KuangzhiGe, 2025-02-23:
I attempted to run the CogACT evaluation script within the SimplerENV environment, but my script appeared to hang during the creation of the ManiSkill2 environment. I attempted to log the program’s execution flow and ultimately discovered that it stalls at the following line:

```python
env = build_maniskill2_env(
    env_name,
    **additional_env_build_kwargs,
    **kwargs,
)
```

Below is the relevant output during runtime:
```
[DEBUG] Will call gym.make() with env_name=GraspSingleOpenedCokeCanInScene-v0, kwargs={'lr_switch': True, 'urdf_version': 'None', 'obs_mode': 'rgbd', 'robot': 'google_robot_static', 'sim_freq': 513, 'control_mode': 'arm_pd_ee_delta_pose_align_interpolate_by_planner_gripper_pd_joint_target_delta_pos_interpolate_by_planner', 'control_freq': 3, 'max_episode_steps': 80, 'scene_name': 'google_pick_coke_can_1_v4', 'camera_cfgs': {'add_segmentation': True}, 'rgb_overlay_path': './ManiSkill2_real2sim/data/real_inpainting/google_coke_can_real_eval_1.png', 'rgb_overlay_cameras': ['overhead_camera']}
[2025-02-23 16:31:31.571] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2025-02-23 16:31:31.572] [svulkan2] [warning] Continue without GLFW.
```

After this point, the program remains stuck without producing any additional output.

I would greatly appreciate any insights you can provide, and I am thankful for your contributions!

xuanlinli17, 2025-02-25:
Does the troubleshooting section of readme solve it?

KuangzhiGe, 2025-02-25:
I re-installed the virtual environment and the problem is solved. Thanks for your reply!

closed 2025-02-25 by KuangzhiGe

LukeLIN-web, 2025-05-02:
> I attempted to run the CogACT evaluation script within the SimplerENV environment, but my script appeared to hang during the creation of the ManiSkill2 environment. I attempted to log the program’s execution flow and ultimately discovered that it stalls at the following line:
> 
> env = build_maniskill2_env(
>     env_name,
>     **additional_env_build_kwargs,
>     **kwargs,
> )
> Below is the relevant output during runtime:
> 
> ```
> [DEBUG] Will call gym.make() with env_name=GraspSingleOpenedCokeCanInScene-v0, kwargs={'lr_switch': True, 'urdf_version': 'None', 'obs_mode': 'rgbd', 'robot': 'google_robot_static', 'sim_freq': 513, 'control_mode': 'arm_pd_ee_delta_pose_align_interpolate_by_planner_gripper_pd_joint_target_delta_pos_interpolate_by_planner', 'control_freq': 3, 'max_episode_steps': 80, 'scene_name': 'google_pick_coke_can_1_v4', 'camera_cfgs': {'add_segmentation': True}, 'rgb_overlay_path': './ManiSkill2_real2sim/data/real_inpainting/google_coke_can_real_eval_1.png', 'rgb_overlay_cameras': ['overhead_camera']}
> [2025-02-23 16:31:31.571] [svulkan2] [error] GLFW error: X11: Failed to open display 
> [2025-02-23 16:31:31.572] [svulkan2] [warning] Continue without GLFW.
> ```
> 
> After this point, the program remains stuck without producing any additional output.
> 
> I would greatly appreciate any insights you can provide, and I am thankful for your contributions!

Same error,  reinstall virtual env but still  cannot work .