simpler-env/SimplerEnv #73: RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed on A100 (when using other gpus except CUDA_VISIBLE_DEVICES=0)

Liang-ZX, 2025-02-25:
Hi authors,

Thanks for your wonderful work. But I met a runtime error when I run
```bash
CUDA_VISIBLE_DEVICES=4 python simpler_env/simple_inference_visual_matching_prepackaged_envs.py --policy rt1 \
        --ckpt-path ./checkpoints/rt_1_tf_trained_for_000400120  --task google_robot_pick_coke_can  \
        --logging-root ./results_simple_eval/  --n-trajs 10
```
It only works when I run it on `CUDA_VISIBLE_DEVICES=0`. How can I fix it? Thank you.

The `vulkaninfo` is shown below.
[image]

xuanlinli17, 2025-02-26:
Try `DISPLAY="" CUDA_VISIBLE_DEVICES=x python xxx`

LukeLIN-web, 2025-05-02:
> Hi authors,
> 
> Thanks for your wonderful work. But I met a runtime error when I run
> 
> CUDA_VISIBLE_DEVICES=4 python simpler_env/simple_inference_visual_matching_prepackaged_envs.py --policy rt1 \
>         --ckpt-path ./checkpoints/rt_1_tf_trained_for_000400120  --task google_robot_pick_coke_can  \
>         --logging-root ./results_simple_eval/  --n-trajs 10
> It only works when I run it on `CUDA_VISIBLE_DEVICES=0`. How can I fix it? Thank you.
> 
> The `vulkaninfo` is shown below. [image]

Did you find the solution?

rakhimovv, 2026-09-01:
In case it helps someone landing here: I hit a symptom in this family on a 4×H100 box, and it turned out to be per-card rather than per-flag.

Two of the four GPUs there fail SAPIEN rendering however they are selected; the other two are fine. It was quiet about it — every call up to and including `take_picture()` succeeds, and the process then hangs in the texture readback at 0% GPU rather than raising (the first thing I saw was `vk::Device::waitForFences: ErrorDeviceLost`). So "it only works on GPU 0" can mean the other cards are unhealthy, not that the flag is wrong.

Might be worth screening each card on its own before assuming it's the selection. A minimal scene is enough:

```python
import sapien.core as sapien
engine = sapien.Engine()
engine.set_renderer(sapien.SapienRenderer(offscreen_only=True))
scene = engine.create_scene()
scene.add_ground(0)
scene.set_ambient_light([.5, .5, .5])
scene.add_directional_light([0, 1, -1], [.5, .5, .5])
cam = scene.add_camera("c", 128, 128, 1.0, 0.1, 100)
cam.set_pose(sapien.Pose([-2, 0, 1]))
scene.step(); scene.update_render(); cam.take_picture()
print("ok", cam.get_float_texture("Color").shape)
```

Run it as `DISPLAY="" CUDA_VISIBLE_DEVICES=<n> timeout 60 python probe.py` once per card. On a good card it prints `ok (128, 128, 4)` in a couple of seconds; on a bad one it times out, so the timeout matters — it hangs instead of erroring.

One thing that did reliably break otherwise-healthy cards for me, in case it is what you are running into: selecting the GPU by passing `renderer_kwargs={"device": "cuda:N"}` rather than by `CUDA_VISIBLE_DEVICES`. With several GPUs visible that splits device selection — Vulkan renders on N while OptiX stays on CUDA device 0 — and the ray-traced envs then fail with `OPTIX_ERROR_INTERNAL_ERROR` before wedging in the same way. The `DISPLAY="" CUDA_VISIBLE_DEVICES=x` suggestion above avoids that entirely.