simpler-env/SimplerEnv #74: Are the tasks in SimplerEnv based on I.I.D. expeirment?

Lucky-Light-Sun, 2025-03-03:
Hello, thanks for your impressive work.
In recent days, I wanna to do some out of distribution tasks for google robot and widowx robot. But I see in simpler paper that "The tasks are chosen to be representative of those in the corresponding training datasets". So I wanna to ask 

1. Is every thing including objects, distractors, backgrounds and instructions already present in training dataset ? Which means that the model training on OXE dataset has already seen these things.
2. Is combination of objects, scenes and so on same in the training datset(maybe just follow some trajectories in training dataset i.i.d).

Looking forward to you reply.

xuanlinli17, 2025-03-03:
1. Instructions are in-distribution. Objects being manipulated are in-distribution. Some distractors are in-distribution and others are ood. For variant aggregation evaluation, backgrounds, table textures, lighting conditions are out-of-distribution. For visual matching evaluation, these are as in-distribution as possible (but still slightly ood).
2. Not necessarily; but our eval is iid. The difference is our eval's iid does not guarantee in-distribution with the training data.

Lucky-Light-Sun, 2025-03-04:
> 1. Instructions are in-distribution. Objects being manipulated are in-distribution. Some distractors are in-distribution and others are ood. For variant aggregation evaluation, backgrounds, table [...]

Thanks for your timely reply. I have some other questions:

1. the object position is set by you or following the training dataset? I see that bridge/widowx evaluation use the `obj_episode_range` as function call paramter.
2. In TABLE IV and TABLE V, the real world evaluation test is evaluated by you, or evaluated in the original paper(however i do not find), or the auther team evaluated it again as SimplerEnv setting?

xuanlinli17, 2025-03-04:
1. (Range) set by me. They are not necessarily in training dataset.
2.  Evaluated by Google for Google Robot and Berkeley & Stanford for Bridge. The real eval object poses roughly match the ones in sim (due to evaluator variances, they are not exact match).

closed 2025-03-05 by Lucky-Light-Sun