simpler-env/SimplerEnv #76: Segmentation Error when load model to device.

LukeLIN-web, 2025-03-08:
`https://github.com/DelinQu/SimplerEnv-OpenVLA/`

I am using numpy 1.26

```
~/SimplerEnv-OpenVLA$ bash scripts/run_openvla.sh 

> self.vla = self.vla.to("cuda:0", dtype=torch.bfloat16)
(Pdb) n
scripts/bridge.sh: line 51: 761268 Segmentation fault      (core dumped) CUDA_VISIBLE_DEVICES=${gpu_id} python simpler_env/main_inference.py --policy-model ${policy_model} --ckpt-path ${ckpt_path} --action-ensemble-temp ${action_ensemble_temp} --logging-dir ${logging_dir} --robot ${robot} --policy-setup widowx_bridge --control-freq 5 --sim-freq 500 --max-episode-steps 120 --env-name PutEggplantInBasketScene-v0 --scene-name ${scene_name} --rgb-overlay-path ${rgb_overlay_path} --robot-init-x ${robot_init_x} ${robot_init_x} 1 --robot-init-y ${robot_init_y} ${robot_init_y} 1 --obj-variation-mode episode --obj-episode-range 0 24 --robot-init-rot-quat-center 0 0 0 1 --robot-init-rot-rpy-range 0 0 1 0 0 1 0 0 1
```

LukeLIN-web, 2025-03-08:
If I don't move to device, I have another problem
```
[2025-03-08 18:08:18.317] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2025-03-08 18:08:18.317] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/homSimplerEnv-OpenVLA/simpler_env/evaluation/maniskill2_evaluator.py", line 89, in run_maniskill2_eval_single_episode
    is_final_subtask = env.is_final_subtask()
AttributeError: 'TimeLimit' object has no attribute 'is_final_subtask'
```

xuanlinli17, 2025-03-09:
What's your gymnasium version?

xuanlinli17, 2025-03-09:
Try `DISPLAY="" CUDA_VISIBLE_DEVICES=x python xxx`
Also what's your gymnasium version?

closed 2025-03-13 by LukeLIN-web

LukeLIN-web, 2025-05-10:
> `https://github.com/DelinQu/SimplerEnv-OpenVLA/`
>
> I am using numpy 1.26
>
> ```
> ~/SimplerEnv-OpenVLA$ bash scripts/run_openvla.sh
>
>
> (Pdb) n
> scripts/bridge.sh: line 51: 761268 Segmentation fault [...]

It happens again and I don't know why