simpler-env/SimplerEnv #77: Source of Real Evaluation Results

unctitus, 2025-03-09:
Hi,

I am wondering where are the [real evaluation results](https://github.com/simpler-env/SimplerEnv/blob/052eaea54dc7c37447cf40c499134b55534bbde4/simpler_env/utils/metrics.py#L7) from `simpler_env/utils/metrics.py` are coming from? 

I want to add more environments and would like to test the real to sim gap in a similar way as the paper. 

Thanks!

xuanlinli17, 2025-03-09:
Google robot eval by Google people; Bridge eval by berkeley and stanford.

unctitus, 2025-03-10:
Thanks for the quick response! Just to clarify, did they conduct real evaluations only for the environments/tasks already in the repository, or are there additional resources with real evaluation results for other environments/tasks?

xuanlinli17, 2025-03-10:
Tasks are the same as in simulation (this repository)

closed 2025-03-19 by xuanlinli17