simpler-env/SimplerEnv #79: ImportError: libvulkan.so.1: cannot open shared object file: No such file or directory

yolo-hyl, 2025-03-14:
My computer GPU is H20 * 8, Ubuntu2204
nvcc -v is cuda-12.8
run the env:
```
Traceback (most recent call last):
  File "/home/weihua/nlp/SimplerEnv/simpler_env/main_inference_magma.py", line 6, in <module>
    from simpler_env.evaluation.argparse import get_args
  File "/home/weihua/nlp/SimplerEnv/simpler_env/__init__.py", line 2, in <module>
    import mani_skill2_real2sim.envs
  File "/home/weihua/nlp/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/__init__.py", line 1, in <module>
    from .custom_scenes import *
  File "/home/weihua/nlp/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/__init__.py", line 1, in <module>
    from . import grasp_single_in_scene
  File "/home/weihua/nlp/SimplerEnv/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/grasp_single_in_scene.py", line 5, in <module>
    import sapien.core as sapien
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/sapien/__init__.py", line 9, in <module>
    from sapien import core, sensor, asset, example, utils
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/sapien/core/__init__.py", line 3, in <module>
    from .pysapien import *
ImportError: libvulkan.so.1: cannot open shared object file: No such file or directory
```
Is my machine capable of running this project?
I searched and found that it might be a problem with the `opencv` library
and need run:
`apt install libgl1`
But libgl1 is already the newest version.
Does anyone have the same experience?

yolo-hyl, 2025-03-14:
`apt install vulkan-tools` can solve this error.

but has new error:

```
Traceback (most recent call last):
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/mediapy/__init__.py", line 1753, in write_video
    writer.add_image(image)
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/mediapy/__init__.py", line 1657, in add_image
    if stdin.write(data) != len(data):
BrokenPipeError: [Errno 32] Broken pipe

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/weihua/nlp/SimplerEnv/simpler_env/main_inference_magma.py", line 68, in <module>
    success_arr = maniskill2_evaluator(model, args)
  File "/home/weihua/nlp/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 215, in maniskill2_evaluator
    success_arr.append(run_maniskill2_eval_single_episode(obj_episode_id=obj_episode_id, **kwargs))
  File "/home/weihua/nlp/SimplerEnv/simpler_env/evaluation/maniskill2_evaluator.py", line 163, in run_maniskill2_eval_single_episode
    write_video(video_path, images, fps=5)
  File "/home/weihua/nlp/SimplerEnv/simpler_env/utils/visualization.py", line 28, in write_video
    media.write_video(path, images_npy, fps=fps)
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/mediapy/__init__.py", line 1751, in write_video
    with VideoWriter(path, shape=shape, dtype=dtype, **kwargs) as writer:
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/mediapy/__init__.py", line 1618, in __exit__
    self.close()
  File "/home/weihua/.conda/envs/magma/lib/python3.10/site-packages/mediapy/__init__.py", line 1675, in close
    raise RuntimeError(f"Error writing '{self.path}': {s}")
RuntimeError: Error writing 'results/Magma-8B/bridge_table_1_v1/arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos/PutCarrotOnPlateInScene-v0/rob_0.147_0.028_rot_0.000_-0.000_3.142_rgb_overlay_bridge_real_eval_1/success_obj_episode_0_moved_correct_obj_True_moved_wrong_obj_False_is_src_obj_grasped_True_consecutive_grasp_True_src_on_target_True.mp4': [h264_nvenc @ 0x5582ab958c60] OpenEncodeSessionEx failed: unsupported device (2)
[h264_nvenc @ 0x5582ab958c60] No NVENC capable devices found
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
```
Looks like a lack of supported hardware accelerated encoder (NVENC), but run the commend:
`ffmpeg -encoders | grep nvenc`
output:
```
ffmpeg version 3.4.4 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --enable-ffplay --enable-ffserver
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
 V..... h264_nvenc           NVIDIA NVENC H.264 encoder (codec h264)
 V..... nvenc                NVIDIA NVENC H.264 encoder (codec h264)
 V..... nvenc_h264           NVIDIA NVENC H.264 encoder (codec h264)
 V..... nvenc_hevc           NVIDIA NVENC hevc encoder (codec hevc)
 V..... hevc_nvenc           NVIDIA NVENC hevc encoder (codec hevc)
```

cacard, 2025-06-06:
my env is:

4090 + windows + wsl + docker
ldconfig -p | grep nvidia can found_64 :
libnvidia-encode.so => /lib/x86 .....

but error when use ffmpeg push rtmp:
OpenEncodeSessionEx failed: unsupported device (2): details
No capable devices found

tell me why :-(