simpler-env/SimplerEnv #83: Any speed up method since only run `  pick_coke_can_variant_agg.sh` takes 40 mins

LukeLIN-web, 2025-04-11:
I don't use all  coke_can_options_arr `# declare -a coke_can_options_arr=("lr_switch=True" "upright=True" "laid_vertically=True") `
only  `declare -a coke_can_options_arr=("upright=True")` 
But it still takes 40 mins. Is there any speed up method?

I try to only run  first block 
`scene_name=google_pick_coke_can_1_v4` and don't run the others like `declare -a scene_arr=("Baked_sc1_staging_objaverse_cabinet1_h870"` But i cannot get `standing sim variant avg success `
how can i get `standing sim variant avg success ` when I don't run all blocks?

xuanlinli17, 2025-04-11:
It's quite strange. Looks like GPU might be unused during eval. What GPU are you using and what's gpu utilization?

LukeLIN-web, 2025-04-12:
> It's quite strange. Looks like GPU might be unused during eval (pure cpu eval is very slow). What GPU are you using and what's gpu utilization?
> 
> You need to keep all scenes to get `standing sim variant avg success`.

Hi, it is ` Compute   0%  16214MiB  17%     0%   3870MiB python simpler_env/main_inference.py `  `  Graphic  11%  16214MiB  17%     0%   3931MiB python simpler_env/main_inference.py --policy-model `  Maybe they are not use it  not use to compute.  I am using H100.

xuanlinli17, 2025-04-12:
Do you have the correct tensorflow version? 
```
pip install tensorflow==2.15.0
pip install -r requirements_full_install.txt
pip install tensorflow[and-cuda]==2.15.1 # tensorflow gpu support
```

xuanlinli17, 2025-04-12:
Also if you are evaluating Octo, see https://github.com/simpler-env/SimplerEnv?tab=readme-ov-file#octo-inference-setup

xuanlinli17, 2025-04-12:
If model is not using GPU, a warning will be raised.

LukeLIN-web, 2025-04-12:
> Do you have the correct tensorflow version?
> 
> ```
> pip install tensorflow==2.15.0
> pip install -r requirements_full_install.txt
> pip install tensorflow[and-cuda]==2.15.1 # tensorflow gpu support
> ```

Thank you! Did you have a advice for torch version? I am face this problem https://github.com/simpler-env/SimplerEnv/issues/30#issuecomment-2799036369

xuanlinli17, 2025-04-12:
I don't think any part of the existing code uses torch? If so you can just insteall cpu torch I think. I locally have torch 2.3.1 but I haven't tested it for a long time.

LukeLIN-web, 2025-04-12:
> I don't think any part of the existing code uses torch? If so you can just insteall cpu torch I think. I locally have torch 2.3.1 but I haven't tested it for a long time.

Thank you. torch 2.3.1 works fine for my openvla!  I think it takes long in rendering. `Graphic ` 2-3x time longer than computing.  I will try to not produce videos.

xuanlinli17, 2025-04-12:
All policy take RGB images as input so RGB images are always rendered. If graphics is the problem then I think ffmpeg might be causing the issue. In this case it means there is a lack of system memory (and if you use a cluster, other jobs might be taking too much memory).

xuanlinli17, 2025-04-12:
Generally on 4090 a single episode of pick coke can is like 3 seconds.

xuanlinli17, 2025-04-12:
Pick coke can uses rasterization so there's no ray tracing so env shouldn't be slow on non-RTX gpus; you can try to bench time for each component to see what's the slowest.

LukeLIN-web, 2025-04-12:
> All policy take RGB images as input so RGB images are always rendered. If graphics is the problem then I think ffmpeg might be causing the issue. In this case (for basically all ffmpeg-caused slow video saving issues) it means there is a lack of system memory (and if you use a cluster, other jobs might be taking too much memory).

My memory should be fine:  215GB /756GB.  Did you means there is not option to (doesn't produce videos)?

xuanlinli17, 2025-04-12:
If so then ffmpeg shouldn't be the bottleneck.

LukeLIN-web, 2025-04-12:
Sorry I forget mention an important thing, maybe it is because my every episode  is fail , so it reach the max steps, so it is slow?   

```
        raw_action, action = model.step(image, task_description)
        predicted_actions.append(raw_action)
        predicted_terminated = bool(action["terminate_episode"][0] > 0)
        if predicted_terminated:
            if not is_final_subtask:
                # advance the environment to the next subtask
                predicted_terminated = False
                env.advance_to_next_subtask()
```
takes  Step time: 0.039 

```
        env_start_time = time.time()
        # step the environment
        obs, reward, done, truncated, info = env.step(
            np.concatenate(
                [action["world_vector"], action["rot_axangle"], action["gripper"]]
            ),
        )
        env_end_time = time.time()
        print(f"Env step time: {env_end_time - env_start_time}")
``` takes Env step time: 0.048

```

    video_start_time = time.time()
    for k, v in additional_env_build_kwargs.items():
        env_save_name = env_save_name + f"_{k}_{v}"
    .... 
    video_path = os.path.join(logging_dir, video_path)
    write_video(video_path, images, fps=5)
    video_end_time = time.time()
    print(f"Video write time: {video_end_time - video_start_time}")

    action_start_time = time.time()
    # save action trajectory
    action_path = video_path.replace(".mp4", ".png")
    action_root = os.path.dirname(action_path) + "/actions/"
    os.makedirs(action_root, exist_ok=True)
    action_path = action_root + os.path.basename(action_path)
    model.visualize_epoch(predicted_actions, images, save_path=action_path)
    action_end_time = time.time()
    print(f"Action save time: {action_end_time - action_start_time}")

Video write time: 0.2038569450378418
Action save time: 0.6596884727478027
```

xuanlinli17, 2025-04-12:
0.039*75 < 3s, so I think the policy forward takes ~7s; you can check

LukeLIN-web, 2025-04-12:
The total `    while not (predicted_terminated or truncated):` loops takes 7.29 - 8.37 s. 
   `model.step()` takes 0.039s/step
`env.step(` takes 0.048s/ step

0.039+0.048=0.087 

Total timestep: 80,  0.087*80=6.96

xuanlinli17, 2025-04-12:
But it doesn't add up to 10s?

LukeLIN-web, 2025-04-12:
First prepare part takes 1.76-1.91s
```
    if additional_env_build_kwargs is None:
        additional_env_build_kwargs = {}

    # Create environment
    kwargs = dict(
        obs_mode="rgbd",
        robot=robot_name,
        sim_freq=sim_freq,
        control_mode=control_mode,
        control_freq=control_freq,
        max_episode_steps=max_episode_steps,
        scene_name=scene_name,
        camera_cfgs={"add_segmentation": True},
        rgb_overlay_path=rgb_overlay_path,
    )
    if enable_raytracing:
        ray_tracing_dict = {"shader_dir": "rt"}
        ray_tracing_dict.update(additional_env_build_kwargs)
        # put raytracing dict keys before other keys for compatibility with existing result naming and metric calculation
        additional_env_build_kwargs = ray_tracing_dict
    env = build_maniskill2_env(
        env_name,
        **additional_env_build_kwargs,
        **kwargs,
    )
    # __import__('ipdb').set_trace()
    # initialize environment
    env_reset_options = {
        "robot_init_options": {
            "init_xy": np.array([robot_init_x, robot_init_y]),
            "init_rot_quat": robot_init_quat,
        }
    }
    if obj_init_x is not None:
        assert obj_init_y is not None
        obj_variation_mode = "xy"
        env_reset_options["obj_init_options"] = {
            "init_xy": np.array([obj_init_x, obj_init_y]),
        }
    else:
        assert obj_episode_id is not None
        obj_variation_mode = "episode"
        env_reset_options["obj_init_options"] = {
            "episode_id": obj_episode_id,
        }
    obs, _ = env.reset(options=env_reset_options)
    # for long-horizon environments, we check if the current subtask is the final subtask
    is_final_subtask = env.unwrapped.is_final_subtask()

    # Obtain language instruction
    if instruction is not None:
        task_description = instruction
    else:
        # get default language instruction
        task_description = env.get_language_instruction()
    print(task_description)

    # Initialize logging
    image = get_image_from_maniskill2_obs_dict(env, obs, camera_name=obs_camera_name) #可以用wrist吗? 
    images = [image]
    predicted_actions = []
    predicted_terminated, done, truncated = False, False, False

    # Initialize model
    model.reset(task_description)

    timestep = 0
    success = "failure"
``` 

Second part, The total    ` while not (predicted_terminated or truncated):` loops takes 7.29 - 8.37 s. 

third part,  save video and action  takes   around 0.85 s

so around  1.76+7.5+0.85=10.11

xuanlinli17, 2025-04-12:
Yeah for the same env the 1.76s can be saved by not re-creating the env and just reset the env w/ different robot & object pose ; but the majority of time is still 50% policy forward and 50% env step

LukeLIN-web, 2025-04-13:
> Yeah for the same env the 1.76s can be saved by not re-creating the env and just reset the env w/ different robot & object pose ; but the majority of time is still 50% policy forward and 50% env step

Thank you for your time! I will keep thinking

xuanlinli17, 2025-04-13:
Essentially the way to speed up both model inference & env is via parallelizing envs; it's already done in ManiSkill3 for widowx envs but not yet for google robot envs.

xuanlinli17, 2025-04-16:
Yes, you can modify the evaluation code to return early.

jasper0314-huang, 2025-04-16:
Got it. Thanks!

LukeLIN-web, 2025-04-16:
> Hi [@xuanlinli17](https://github.com/xuanlinli17) [@LukeLIN-web](https://github.com/LukeLIN-web),  你好 I have a question that might be stupid, can we return early from an episode once success has been achieved?我有一个可能很愚蠢的问题，一旦成功，我们可以提前从发作中回来吗？
> 
> For example, if `success` becomes `True` at the 12th step, all the remaining steps seem unnecessary.例如，如果 `success` 在第 12 步变为 `True`，则其余所有步骤似乎都是不必要的。 Would it be okay to just break out of the loop at that point?在那个时候跳出循环可以吗？
> 
> [image]

great idea!

LukeLIN-web, 2025-04-18:
> save video and action takes around 0.85 s

And we have to save video otherwise cannot count metrics now .

Boltzmachine, 2025-06-09:
Is it ok to parallelize it? I am worried that there are some issues when running multiple instances on a machine. (I have experience of other softwares that if you run multiple instances there are bugs)

xuanlinli17, 2025-06-09:
Bridge envs are parallelized in ManiSkill3. Google Robot envs tbd.