simpler-env/SimplerEnv #84: `main_inference.py: error: argument --enable-raytracing: expected one argument`

LukeLIN-web, 2025-04-14:
When I try to run `bash scripts/run_openvla.sh`. 
is it should be true or false?

xuanlinli17, 2025-04-15:
See https://github.com/simpler-env/SimplerEnv/blob/main/scripts/rt1_drawer_visual_matching.sh
It should be `--enable-raytracing` as it stores true. I don't have `run_openvla.sh` in my repo, so it's a fork.

LukeLIN-web, 2025-04-16:
> See https://github.com/simpler-env/SimplerEnv/blob/main/scripts/rt1_drawer_visual_matching.sh It should be `--enable-raytracing` as it stores true. I don't have `run_openvla.sh` in my repo, so it's a fork.

Yes I am runnning `  drawer_variant_agg.sh`  and show this error.

What is the effect/benefit of enable-raytracing?   It will make the A100 GPU evaluate slower. I saw `enable raytracing, slow for non RTX gpus`

xuanlinli17, 2025-04-16:
Unfortunately for drawer envs (non-table-top envs) you have to use ray tracing, otherwise there is a large visual gap; A100 is not RTX GPU so it's a lot slower than RTX GPUs.

closed 2025-04-17 by LukeLIN-web