simpler-env/SimplerEnv #89: Hang when creating env

LukeLIN-web, 2025-05-01:
Before it can run , but have err`RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent after eval,`

`it is same with you . The error occurs after running 10 episodes on all policies`

https://github.com/simpler-env/SimplerEnv/issues/68 

then I try to fix it , I  install `vulkan-tools`  , I run `bridge.sh `, 

[2025-05-01 14:18:01.329] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2025-05-01 14:18:01.329] [svulkan2] [warning] Continue without GLFW.

then the program  hang here 
```
   env = build_maniskill2_env(
        env_name,
        **additional_env_build_kwargs,
        **kwargs,
    )
    # initialize environment
    env_reset_options = {
        "robot_init_options": {
            "init_xy": np.array([robot_init_x, robot_init_y]),
            "init_rot_quat": robot_init_quat,
        }
    }
    if obj_init_x is not None:
        assert obj_init_y is not None
        obj_variation_mode = "xy"
        env_reset_options["obj_init_options"] = {
            "init_xy": np.array([obj_init_x, obj_init_y]),
        }
    else:
        assert obj_episode_id is not None
        obj_variation_mode = "episode"
        env_reset_options["obj_init_options"] = {
            "episode_id": obj_episode_id,
        }
    obs, _ = env.reset(options=env_reset_options)
```
and cannot be shut down

StoneT2000, 2025-05-02:
What branch of simpler are you on?

StoneT2000, 2025-05-02:
And what is bridge.sh? I don't see this file.

LukeLIN-web, 2025-05-02:
> What branch of simpler are you on?

I am on a forked branch  https://github.com/DelinQu/SimplerEnv-OpenVLA

StoneT2000, 2025-05-02:
Sorry this might be harder to debug but I can try. Are you using maniskill3 or maniskill 2? I notice you are using sapien 3 but the code is for maniskill 2, which is incompatible

LukeLIN-web, 2025-05-02:
> Sorry this might be harder to debug since its not the original repo but I can try. Are you using maniskill3 or maniskill 2? I notice you are using sapien 3 but the code is for maniskill 2, which is incompatible

Thank you ! I am using maniskill 2, so you means I need to change to sapien 2? How to do it. I install everything from readme.md , but still have this problem.

LukeLIN-web, 2025-05-02:
> Sorry this might be harder to debug since its not the original repo but I can try. Are you using maniskill3 or maniskill 2? I notice you are using sapien 3 but the code is for maniskill 2, which is incompatible

I downgrade to sapien                       2.2.2 but still same problem.

LukeLIN-web, 2025-05-02:
> Sorry this might be harder to debug since its not the original repo but I can try. Are you using maniskill3 or maniskill 2? I notice you are using sapien 3 but the code is for maniskill 2, which is incompatible

I am also trying maniskill 3 ,   `TypeError: BaseEnv.__init__() got an unexpected keyword argument 'robot' was raised from the environment creator for PutCarrotOnPlateInScene-v1 with kwar `  could you have a look ? https://github.com/simpler-env/SimplerEnv/issues/90#issue-3035005224

StoneT2000, 2025-05-02:
Do you want to use the gpu sim of simpler or are you okay with the CPU parallelization?

Debugging is a bit different for both since they use different simulator versions

LukeLIN-web, 2025-05-02:
> Do you want to use the gpu sim of simpler or are you okay with the CPU parallelization?
> 
> Debugging is a bit different for both since they use different simulator versions

I am okay with the CPU parallelization

StoneT2000, 2025-05-02:
Can you try running the original simpler env code? And make sure to not include ipdb

LukeLIN-web, 2025-05-03:
> Can you try running the original simpler env code? And make sure to not include ipdb

I tested my forked code and originial simpler env code
```
import simpler_env
from simpler_env.utils.env.observation_utils import get_image_from_maniskill2_obs_dict

env = simpler_env.make('google_robot_pick_coke_can')
```
This would not hang.

StoneT2000, 2025-05-17:
It is difficult to say what the issue is as that code is running other things not in the original codebase. I recommend trying to remove things one at a time / add one at a time to the code to debug. There are too many changes to see for me to figure it out at the moment.

LukeLIN-web, 2025-05-20:
> It is difficult to say what the issue is as that code is running other things not in the original codebase. I recommend trying to remove things one at a time / add one at a time to the code to debug. There are too many changes to see for me to figure it out at the moment.

Thank you, I wonder if maniskill3 is ready, if so I want to have a try with that branch.

StoneT2000, 2025-05-20:
There is a maniskill 3 branch for this repo, you can try that