simpler-env/SimplerEnv #98: Interface for Default Frame-Wise Patch Annotations in SImplerEnv

MstarLioning, 2025-05-20:
Dear Authors,

Hello! Could you please let me know if SImplerEnv provides an interface that, for each frame, automatically annotates which patches contain the target object and the robotic gripper? I look forward to your reply!

xuanlinli17, 2025-05-20:
Could you elaborate what "patches" is?

MstarLioning, 2025-05-21:
@xuanlinli17 Sorry, I apologize for not explaining myself clearly. What I mean is: in the full image, will there be a bounding box or boundary around the regions containing the target object and the robot that can be directly viewed or accessed?

xuanlinli17, 2025-05-21:
Then you need to render the segmentation mask in the camera. You need to modify the camera config in robot agent to include segmentation https://github.com/haosulab/ManiSkill/blob/493be36121a9dd06071a57172274babe617b789f/mani_skill2/sensors/camera.py#L118

MstarLioning, 2025-05-21:
I can’t thank you enough—thank you for patiently answering my questions!

closed 2025-05-21 by MstarLioning