simpler-env/SimplerEnv #99: About end-effector position API

phandat128, 2025-05-20:
Hi, i am new to simulation environment, can i ask you that: to get the eef_pos of robot, do `obs["agent"]["eef_pos"]` (and `env.unwrapped.agent.robot.get_qpos()` return the same outputs? 
Thank you so much!

xuanlinli17, 2025-05-20:
No they don't;    end-effector 6d pose (xyz, quat) can be obtained via `env.unwrapped.agent.robot.tcp.pose`;  `get_qpos` is robot's joint qpos instead of end-effector pose.

phandat128, 2025-05-21:
Thank you for your quick response, 
I have tried `env.unwrapped.agent.robot.tcp.pose` but i receive an error: `AttributeError: 'sapien.core.pysapien.Articulation' object has no attribute 'tcp'`. Can you suggest what is wrong here?

xuanlinli17, 2025-05-21:
Sorry; see https://github.com/simpler-env/ManiSkill2_real2sim/blob/ef7a4d4fdf4b69f2c2154db5b15b9ac8dfe10682/mani_skill2_real2sim/examples/demo_manual_control_custom_envs.py#L487

phandat128, 2025-05-21:
Thank you, i will try it

closed 2025-05-21 by phandat128