simpler-env/SimplerEnv #106: Missing GroundingTruth entries for selected google_robot_* tasks in Fractal dataset

MstarLioning, 2025-05-29:
Hello,We have selected the following tasks in `SImplerenv`:

```python
[
    "google_robot_pick_coke_can",
    "google_robot_pick_horizontal_coke_can", 
    "google_robot_pick_vertical_coke_can",
    "google_robot_pick_standing_coke_can",
    "google_robot_pick_object",
    "google_robot_move_near_v0",
    "google_robot_move_near_v1",
    "google_robot_move_near"
]
```

We’d like to locate the corresponding episodes (and GroundingTruth annotations) for each of these tasks in the Fractal dataset. However, after searching through the first 15,000 episodes, we only found a GroundingTruth entry for `google_robot_pick_coke_can`. Is this expected? Have we overlooked something?

Our extraction method is as follows:

```python
# Assign episode_id and extract
episode_data = self._extract_episode_data(episode, matched_task, task_counters[matched_task])

# In _extract_episode_data we assemble the actual action trajectory:
action_trajectory = np.array([
    np.concatenate([
        step['action']['world_vector'].numpy(),
        step['action']['rotation_delta'].numpy(),
        step['action']['gripper_closedness_action'].numpy()
    ])
    for step in episode['steps']
])
```

Any pointers on whether the other tasks should appear under different names, or if there’s an alternative way to locate their GroundingTruth, would be greatly appreciated. Thank you!

xuanlinli17, 2025-05-29:
Could you elaborate what "ground truth" means? The names of the mp4 files saved for visualization have the success conditions.

MstarLioning, 2025-06-12:
@xuanlinli17 Sorry to bother!I means the samples in SImplerenv could be easily mapped to a trajectory in open‑x‑embodiment?