simpler-env/SimplerEnv #112: An error occurred on the headless server

hamondyan, 2025-07-26:
Hi, 
First of all, thank you for your contribution to this article on spatialVLA.
I encountered an error when configuring simpler_env on the headless server:

                 sapien.SapienRenderer(offscreen_only=True,**renderer_kwargs)
                 RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent

I've tried many solutions, but none of them work. I would like to know if you have any experience running a successful application on a headless server?

xuanlinli17, 2025-07-26:
Hi, have you tried the troubleshooting section in readme?

YantaiYang-05, 2025-07-29:
I encountered the same error,have you solved it？

xuanlinli17, 2025-07-29:
Try `CUDA_VISIBLE_DEVICES=xx DISPLAY="" python xxx`?

Boltzmachine, 2025-08-07:
This error does not appear on other environments but appears only on move_near_variant_agg